df8oe/UHSDR #1638: Need Tx IQ adjustment at all ham bands

satoryboy, 2018-12-03:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

We really need soft Tx IQ adjustment at every ham band. IMHO a little TX power is not reason to ignore the advantage of digital technology. Quality of Tx signal is very important even on QRP. And it is enough simple, ist't it?

df8oe, 2018-12-03:
No, it isn't. We do need storage place for every band. So we must

1) change layout of band specific storage blocks which would break everyones settings and throw him back to initial start

or

2) create a new block which only holds TX IQ adjustment.

We do want to add some more bandpecific settings to band storage places and create memories for fast access of frequencies/modes/power etc. But we need to discuss what is neccessary to store and what not. All this does have a low priority actually - it eats much time. Additionally nearly nobody would do a complete IQ adjustment (too much work) so maths do their job very good. I never have had missed anything - if you follow the rules that this is "a QRP rig". Most "bad signal issues" do not have their source at missing bandspecific TX IQ adjustments but in hardware reasons and firmware cannot improve anything there...

satoryboy, 2018-12-03:
Yes, in hardware reasons, quadrature driver on 74AC74 - about 4 grad of mistake on high bands. And now we have 2 ways - experiments with hardware or compromise in quality of tx signal, for example, between 14 and 28 MHz. Why? IMHO it's more important problem, than, for example, FreeDV.
Yes, a new block which only holds TX IQ adjustment. Classic scheme - for novice (as now) and expert, - "Adj. Tx IQ on all bands -> ON". Half a day for work to good coder. It's not just my wish, really. Many hams write about it on some forums. Thanks for your answer.

DD4WH, 2018-12-03:
Hi Satoryboy,

if you can do that in half a day, go ahead and program it and make a pull request!

I am a little suspicious about your statements:
"We really need . . . "
"4 grad of mistake . . . "

I am pretty sure you are aware of this:
http://www.panoradio-sdr.de/wp-content/uploads/2017/01/ssb_suppression-1024x614.png
So, 4 degrees of error would mean less than 25dB of opposite sideband suppression!?

Do you have real measured data to support your statement of less than 25dB sideband suppression in your transceiver using UHSDR? Which hardware do you use? Is it the Chinese clone? 

73 Frank DD4WH

satoryboy, 2018-12-03:
Sorry, iteration on 20m and higher do a job not ideally. That's fact. Quote from cqham.ru forum: "Up to 40 dB it was possible to squeeze out, but there is one problem, on some single RF range. If I am balancing on the top ten, the suppression of the mirror is less on fiftene. At the twenty tolerated. That is why I am balancing by 21 megahertz and essentially score on the top ten since there activity is very low. Of course, it would be better if the menu on each band had the opportunity to twist side suppression. But this is a dream. Unfortunately not feasible."

DD4WH, 2018-12-03:
Sorry, satoryboy,

"facts" derived from a forum . . . not even knowing the hardware, the software version etc. . . . 

I prefer measurement data that is properly presented. It is not even clear if IQ adjustment is relevant here at all.

73 Frank DD4WH

df8oe, 2018-12-03:
I have just tested on 3 mcHFs. If properly adjusted at 80m and 10m the suppression of the unwanted mirror is better 55dB. At the worst band it is just 5dB below that. Absolutely ok for a small QRP rig. I do not see any need to adjust at every band (and at 15m and 10m more times). If anybody does want to implement that: go on :) Maybe you do have "tested" on Chinese clones? These do use very cheap inductors and capacitors which have a much worse result...

satoryboy, 2018-12-03:
I have better 43dB on all bands on my device, but not sure it is enough. Exuse me please for disturb and thanks for your time.

df8oe, 2018-12-03:
Better 43dB is sufficient - but not very good. I got > 50dB on all bands. May be low Q capacitors or inductors... McHF is a low budget homebrew rig and > 50dB is nice.

satoryboy, 2018-12-04:
On 80m and 10m (at adj. points) - better 55bB. Minimum of supression (43-45bB) on 21m. Adjusting high point IQ for 21m - 48-50dB without any problem. FR board - V0.6

tinez99, 2018-12-04:
IQ balance on each band is very necessary. Is it bad that there will be more opportunities to get a good signal?
Mathematics is not affected by the layout of boards of different versions. The phase shift is different. Adjustment is needed.

DD4WH, 2018-12-04:
Thanks for your measurements! That is the kind of information we need.

To summarize:

* if we implement adjustment of IQ phase and amplitude for each band, you get for some bands 3 to 5dB more opposite sideband suppression [43-45dB compared to 48-50dB]

So, is it worth it ? (taking into account that only few people will take the time of adjusting / measuring opposite sideband rejection separately for each band)

satoryboy, 2018-12-05:
"if you can do that in half a day, go ahead and program it and make a pull request!"
I will try it. Thanks to SP9BSL!

satoryboy, 2018-12-08:
Please consider my pool-request #1639. Simple decision for a low budget homebrew rig.

df8oe, 2019-02-01:
implemented now.

closed 2019-02-01 by df8oe