df8oe/UHSDR #1644: REQUEST: Display darkening

OK1DSG, 2018-12-11:
Dear OMs,
While using the mchf TCVR in a dark surraundings I would appreciate the possibility to darken the display even more than it is now possible by pressing the Power button.

Do you think it will be possible to add more steps of darkening than three? Or to add one new item into menu for example “Display Intensity” that will set the display intensity between 0-100%. This value could be active for example by selecting the third step of darkening. 

Thank you in advance
Jiri

db4ple, 2018-12-11:
The way the dimming is done, it is not possible to have very low brightness settings because it would create visible flickering of the backlight. Right now, we have a minimal dutycycle of about 25% and 1500/4 = 375 Hz refresh rate. We might be able to go down to about 100 Hz refresh rate. That would reduce the brightness further. Which roughly would get us into 1/15 of the max brightness.

db4ple, 2018-12-17:
Hi,
with little effort we could implement the following duty cycles:
1 (100%), 1/2 (50%), 1/4 (25%), 1/8 (12,5%), 1/16 (6,25%) or
1, 1/3, 1/5, 1/8, 1/11, 1/16 if you would like a little less than halfing of the duty cycle. 
This means essentially twice as many dimming steps and much darker lowest position.

Any objections?

OK1DSG, 2018-12-17:
Hello,
from my side is the first option (1 (100%), 1/2 (50%), 1/4 (25%), 1/8 (12,5%), 1/16 (6,25%) ) fully sufficient.
Thanks a lot!
73!
Jiri

OK1DSG, 2018-12-30:
Dear OMs,
Thank you for solving this issue! It works fine.
PF 2019!
73!
Jiří

closed 2018-12-31 by OK1DSG