df8oe/UHSDR #1645: #1640 Quick Band Scan

wk8s, 2018-12-11:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:
Your bootloader version: 
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

wk8s, 2018-12-11:
v 2.9.86  Pressing and holding the Band+ or Band- does nothing on mcHF. 
Maybe it's not meant to work on mcHF?

db4ple, 2018-12-11:
It is meant to work on all machines. But after giving the code a closer look I doubt that it ever worked. 
In order to change the frequency one has to change df.tune_new to the desired frequency. The code does not do this. So the call will change to the frequency and very soon afterwards will be tuned back to the old frequency since df.tune_new still has the "old" frequency. 

I still have to verify this thought, but it looks that this would be right. In addition there is another small issue in the code. But this is a different  story.

df8oe, 2018-12-15:
fixed.

closed 2018-12-15 by df8oe