df8oe/UHSDR #1648: Sorry, please help

satoryboy, 2018-12-19:
Sorry, please help about pool-request - commit: https://github.com/df8oe/UHSDR/commit/516be8f69868cf862da4456d19f41d22ae068a56?diff=unified

m-chichikalov, 2018-12-19:
Hi Yuri, What's wrong?

satoryboy, 2018-12-19:
OK, I will try to create a new actual branch and commit it.

m-chichikalov, 2018-12-19:
Take a look into "rebase" and "cherry-pick" commands of the git and forget for a while the "revert" you do not need it usually.

In your fork of UHSDR repo I do not see any PULL comit, did you add remove repo as UPSTREAM to do pull?

PS, you can text or call me via Skype if you want rv9yw_max, 73!

m-chichikalov, 2018-12-19:
read this -> https://github.com/df8oe/UHSDR/blob/active-devel/CONTRIBUTING.md

satoryboy, 2018-12-20:
Hope, now it is OK.

m-chichikalov, 2019-01-19:
@df8oe , Andreas, could you close this one. I think it's not relevant anymore.

df8oe, 2019-01-19:
I see this also. @satoryboy : please close your issues if they are no longer relevant.

closed 2019-01-19 by df8oe