df8oe/UHSDR #1650: Just for enhancement - it would be great to be able to write/load config from USB flash disk. Sorry, this work not for me, I'm not good coder.

satoryboy, 2018-12-21:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:
Your bootloader version: 
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

df8oe, 2018-12-21:
This is a very large project. We are facing a solution which writes text config to USB and when reading back parses this text. So you do have a very good compatibility over many versions.

But unfortunately this is a huge task and eats much men power and time. We already do have possibility to read config via USB and small python script to PC and write it back - sufficient in most cases. So actually there is nobody who sees such big progress in USB stick writing that he starts this task...

satoryboy, 2018-12-21:
I see, thank you. Storing/loading from computer is enough for good work with configs.