df8oe/UHSDR #1654: Splitting Graphics Driver Code

db4ple, 2018-12-30:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:
Your bootloader version: 
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

db4ple, 2018-12-30:
A side note: I would think we keep the touchscreen code in the controller side of things for the moment, these are fairly related and the touch screen stuff is not too large.

m-chichikalov, 2018-12-30:
Hi Danilo, this is for me? Take a look my repo if you interested how I do it so far. The major side and idea which I keep in my mind when do it, to broke as much as I can dependencies and make it easy to adopt RTOS later. Right now the graphical part ready for using in concurrency environment. I've changed that transaction start only when first buffer is ready, so there is no delay to prepare buffer.... and others changes, for sure I will prepare some documentation, dont worry.

m-chichikalov, 2018-12-30:
> A side note: I would think we keep the touchscreen code in the controller side of things for the moment, these are fairly related and the touch screen stuff is not too large.

I do not agree with you. It should be separate module and there is nothing related to display's controllers.

m-chichikalov, 2018-12-30:
And BTW, for SPI I achieved improvements around 8-15% for different operations 

12500  --> 11705
--
2991  --> 2148
3332  --> 2531
1016  --> 697
4571 --> 3622
4572 --> 3622
7112 --> 6033
8667 (not correct) --> 3018

and I think to implement Init functions to try both prescalers to be sure that the person this good display using it on higher clock.

I implemented mem-to-mem DMA for F4. Performance increased a bit. going to optimize all print text functions, for now they looks not optimized.

sp9bsl, 2018-12-30:
Danilo, agree. Function names are borrowed from first/old approach of Chris and it looks a bit strange when function name begins with old display name (HY28_...) and we use it for ILI9486 or RA8875. I was thinking about this move a month or two ago...

Maksim, keeping the touchscreen with diplay controller is a key part, so I would not agree with you. The reason is simple: RA8875 has controller for resistive panels inside the display chip and uses the same IO as the display controller, the RPi lcd (more or less frequently used) also shares the same SPI port.

m-chichikalov, 2018-12-30:
That they use common interface is not reason to keep them together. Guys let's start thinking about dependencies, concurrency and stop writing the spaghetti code.

How I managed SPI for LCD and Touch:

First of all I added simple locks mechanism, which can be easely replaced later if we go to RTOS.
locks.c
``` C
/**
 * Semaphore give
 */
void sem_give( volatile bool* sem )
{
	*sem = false;
//	xSemaphoreGiveFromISR( *sem, pdFALSE );
}

/**
 * Semaphore take, may suspend the caller thread
 */
int sem_take( volatile bool* sem, uint timeout )
{
	(void)( timeout );
	// As long as we do not have any concurrency,
	// there is no way that semaphore is already taken by someone.
	// The below implementation for DMA or interrupt driven Functions,
	// they should give back semaphore after it's done.
	if ( !*sem )
	{
		*sem = true;
		return ERR_NONE;
	}
	while ( *sem )
	{
		//FIXME maybe add timeout for more robust code?
		asm("nop");
	}
	*sem = true; // take semaphore after it's released...
	return ERR_NONE;
//	return xSemaphoreTake( *sem, timeout ) ? ERR_NONE : ERR_TIMEOUT;
}
```

Further I added SPI interface - like boundary to separate HW depended code from aplication.
it has the next interface:
``` C
const IO_t SPI_IO = {
	.RequestTransaction = UHSDR_SPI_RequestTransaction,
	.CloseTransaction   = UHSDR_SPI_CloseTransaction,
	.Send_DMA           = UHSDR_SPI_Send_DMA,
	.Send_Polling       = UHSDR_SPI_Send_Polling,
	.Receive_Polling    = UHSDR_SPI_Receive_Polling,
	.Receive_DMA        = UHSDR_SPI_Receive_DMA,
};

const IO_t* UHSDR_SPI_GetIO( void )
{
	return (const IO_t*)&SPI_IO;
}
```
the same for FMC and display driver are using one of them depends on interface it's using.

Go back to SPI display and SPI touch:

Any transaction to the Display should start from taking its semaphore. For now it's just bool type field in Display struct.
By this we are protecting our buffers from corruption when Interrupt or DMA approach are in invoked.

Each interface also has semaphore, so before start transaction over SPI the 
`IO->RequestTransaction ( uint Timeout, void* ExtraParameter ); `
shoud be called.
if used polling function of interface the 
`IO->CloseTransaction();` 
should be called manually
if DMA was used the close transaction would be called automatically from IRQ.

the code you can look at my repo... it's not done yet but more or less...

I also add into SPI interface implementation of
`IO->RequestTransaction ( uint Timeout, void* ExtraParameter ); `
that it takes the struct of used chip select pin, so it set it back automatically when transaction over DMA is done. The main code do not need to handle CS anymore. And does not matter who is calling it touch or display or third party which using the same SPI...

The same done for FMC. Your 8875 display and touch inside it just should concurently share the same interface and respect agrement to take semaphore and give it back.

so, it has not only touch controler inside but also others, for ex. buttons controller, so if I need to use it do you think I will put code at the same module to scan buttons?

m-chichikalov, 2018-12-30:
I also do not like that we injecting everything into module namespace just by including uhsdr_board.h this is not right, I would prefer to split it and include only realy needed dependencies into module.