df8oe/UHSDR #1664: Bug ver 2.11.1

ES8AS, 2019-01-02:
Firmware version: 2.11.1
Bootloader version: 5.0.2
Hardware SDR TRX RS-958B

After upgrade to ver 2.11.1 the screen is blinking during booting and after boot. Transceiver switch off is only via power supply switch off.

db4ple, 2019-01-02:
If the screen is white and the screen is blinking, this is normally the bootloader indicating a failed flash process. Try reflashing the binary. And check the binary was downloaded correctly.

ES8AS, 2019-01-02:
The screen is not white. I can read on the screen everything, but blinking. Flash process completed without errors.

db4ple, 2019-01-02:
Ok. That is a different issue then. 2.11.0 does not have this issue?

ES8AS, 2019-01-02:
Tried to flash once more - result the same - blinking and can't switch off normally. The ver.2.11.0 is OK.

db4ple, 2019-01-02:
OK, then stay on 2.11.0 until further notice.

ES8AS, 2019-01-02:
Ok. Thanks! Happy New Year!

df8oe, 2019-01-02:
I cannot confirm here on any device.

SergiuEpure, 2019-01-02:
Same to me

db4ple, 2019-01-02:
@SergiuEpure: You mean, you have no problems with 2.11.1?

SergiuEpure, 2019-01-02:
@db4ple: Sorry for misunderstanding. Same problem to me, screen blinking and transceiver is off only via power supply switch off.

db4ple, 2019-01-02:
@SergiuEpure: Please report your hardware, especially processor type and size of flash...

db4ple, 2019-01-02:
All of you, please also try 2.11.2 (it should not change the situation, but at least give it a try).

SergiuEpure, 2019-01-02:
@db4ple my procesor is stm32f439, 2048kb flash and 256kb ram, boar if i remember well is v4

db4ple, 2019-01-02:
Thanks. @ES8AS: Same information required from you. Also display connection is of interest (SPI or Parallel), also in system info menu.

ES8AS, 2019-01-02:
From system menu: display HY28A/B Para, disp controller ILI9325, SI570/56.32Mhz, SI351A not found, EPROM 24xx1025/128K, FLASH 1024kB, RAM 192kB, RF Board - mcHF RF Board (ver?)

db4ple, 2019-01-02:
Ok, I found the issue. No need to give me further information, I think. I pushed the fix. Once 2.11.2 is released, please test. The issue was only affecting STM32F4 users

ES8AS, 2019-01-02:
The CPU is stm32f407VGT6 1M memory (as described on eBay)

db4ple, 2019-01-02:
Thanks to all of you for reporting and helping to fix the issue!

ES8AS, 2019-01-02:
You are always welcome!

commit 053fe450 (2019-01-02): Merge pull request #1666 from db4ple/24bit

Fixed #1664 and a warning...
closed 2019-01-02 by df8oe