df8oe/UHSDR #1670: cycling thru dsp gets stuck on peak

regor60, 2019-01-04:
Your firmware version: 2.10.0
Your bootloader version: 5.0.2

### Hardware 
 * UI Board: RS-978 clone
 * RF Board: mcHF 512k CPU

### Describe the issue:
Cycling thru NR using DSP button gets stuck on Peak option and will not cycle around.  Radio has to be rebooted to allow other options to be selected once again. Not a stuck button

df8oe, 2019-01-04:
I have tested on three different mcHFs - I cannot confirm. Please attach your configuration possibly it is due to a setting in your config.

db4ple, 2019-01-14:
Hi @regor60,
you still have the problem? And if so, did you try a newer firmware? Maybe you disabled some modes via the virtual DSP buttons (via touchscreen). Please check this.

regor60, 2019-01-14:
Yes still the problem.  I will check

df8oe, 2019-02-01:
Problem does exist? If not, please close issue. If so, please provide your settings file for debug purposes.

regor60, 2019-02-01:
Problem persists.  How do I provide a settings file?

df8oe, 2019-02-01:
Follow this guide:
https://github.com/df8oe/UHSDR/wiki/Backup-and-Restore#backup--restore-on-external-computer-since-2919

Sorry - noone else seems to have this issue. If you cannot provide youzr settings for analysis we will be unable to solve this.

df8oe, 2019-02-01:
Sorry - not reproduceable by us nor anyone else has reported this issue. Without accessing your config noch chance to fix anything. Hint: try to start with a clean default config and see if the problem persists. If not, it is a setting issue at your rig. If it persists it is a hardware issue I think.