df8oe/UHSDR #1682: Refactoring: PA Level Control and Display

db4ple, 2019-01-09:
See and comment this change  I am working on in my github repo:
https://github.com/db4ple/UHSDR/tree/pa_level_control

Please comment, I'll plan to finish this soon.

Objectives: 
- better separation of (generic) UI and RF HW handling.
- better (user) support for common issues related to low or missing power output. 

- tried to separate the PA power steps (which are more a UI thing)
from the calculation of the settings to control power output by the PA.
This makes it easier to connect different PA (with differing power levels)
using the same code (which is at the moment basically the calculation of a gain value to
interface the DSP output with the PA side of things).
- Also changed display code to dynamically calculate the power display.
Range 0mW to 100W will fit into existing box.

df8oe, 2019-01-09:
Nice. From my view you get a GO.

sp9bsl, 2019-01-09:
I also vote GO.

db4ple, 2019-03-20:
It is integrated since a while, has now been extended by more messages if no power output will be generated...

closed 2019-03-20 by db4ple