df8oe/UHSDR #1691: Error during compile starting with Version 2.11.xx

DL8EBD, 2019-01-16:
## Issue Data goes here (please remove text above):

Your firmware version: 2.11.xx
Your bootloader version: 5.0.3

Windows10
Eclipse v4.4.2-20180930-2018-09/20180930-0922-gnumcueclipse-4.4.2-2018-09-win32.win32.x86
ARM Toolchain 7-2018q2

### Hardware 
SParrow

### Describe the issue:
Hi folks,
since last weekend I can't compile the versions 2.11.xx and higher without errors!
I get 6 errors which all refer to "UHSDR_canary" ???
Versions up to 2.10.xx run without errors.
To build the DebugLib’s is no problem!
but the bin’s do not finish without errors:

[image]

any ideas?
more information needed?

thx
Thomas-DL8EBD

db4ple, 2019-01-16:
Fixed in last PR #1690

DL8EBD, 2019-01-16:
Andreas, Danilo
works again as usual!
Can be closed

thanks and greetings
Thomas-DL8EBD

closed 2019-01-16 by DL8EBD