df8oe/UHSDR #1692: Board Selection for McHF or OVI40 breaks Wiki build instructions.

rensd, 2019-01-16:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version: 2.11.12
Your bootloader version: 5.0.3
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.6
 * RF Board: mcHF 0.6

### Describe the issue:

The building instruction in the Wiki no longer work.
Board selection is now part of the (g)make command and is done
in .travis.yml but that won't work if you just cd into mchf-eclipse 
for building

Update to:
cd mchf-eclipse 
make CONFIGFLAGS="-DUI_BRD_MCHF -DRF_BRD_MCHF" all

Should fix that

73's Rens | PA3AXA

db4ple, 2019-01-16:
Hi Rens,
thanks for the note. But why not fix it yourself and get the full fame for it? If you can write an issue here, you can also fix the Wiki ;-)

m-chichikalov, 2019-01-16:
This Issue on me, so far... 
@db4ple  As I understand we do not want to specify RF board? Should I change make and travis to build default MCHF_UI if not mention like "make all"?

db4ple, 2019-01-16:
@m-chichikalov: I know. And next time we should keep in mind to update instructions. 

I think a simple default config for the build targets would solve this problem. Since most users do have a mcHF this is quite a safe default. Nevertheless we have to document the main idea of how to build other device targets. Not everyone knows that .travis.yml is the place to look, let alone is able  to understand travis.yml (or our makefile) without struggle.

df8oe, 2019-01-16:
My GO for "mchf rf default". Actually we do not need special builds and complete hardware of rf board is done by lifetime identification. Maybe that will change in the future - so it is good to have a mech for that already present.

vy 73
Andreas

m-chichikalov, 2019-01-16:
@df8oe , sorry Andreas, I did not get your point? Do you mean as long as we identify the RF board at the RUN time, there is no needs to use extra -D defines for it? If so, I agree. Will remove it later.

db4ple, 2019-01-16:
I don't understand what "mchf rf default" is supposed to mean?
I think we are overcomplicating things here. Just set at the beginning of the Makefile the CONFIGFLAGS variable to "-DUI_BRD_MCHF -DRF_BRD_MCHF" if no CONFLIGFLAGS are set, would do the job sufficiently (in addition to a pointer where to get more information). With this the old
make all etc. will work 100% as before.

m-chichikalov, 2019-01-21:
Might be closed.

closed 2019-01-23 by df8oe