df8oe/UHSDR #1706: RTTY/CW Decode Line

DF5LI, 2019-01-22:
Firmware version: 2.11.18
Bootloader version: 5.0.3

 * UI Board: mcHF 0.x / OVI40 1.x (all, H7 +F7)
 * RF Board: mcHF 0.x / OVI40 1.x (all)

With the version 2.11.17 unfortunately the rtty-decode line was gone. But with version 2.11.18 it works again. Only the small > character at the beginning of the line remains missing. Bug or feature?

73 Harri  DF5LI

m-chichikalov, 2019-01-23:
Hi all, Honestly and personally I would like assign < & > chars with increasing and decreasing CW speed in macros and come up with another way to indicate rx and tx. I'm thinking about implementing indication about RX and TX text in different colors, also about the indication what was transmitted already and what not. Would like to hear you crazy ideas what would you like to have regarding macros, cw, and paddling from USB keyboard. Thanks

73! Max RV9YW

DF5LI, 2019-01-23:
Hi Max! I actually missed only the small > sign at the beginning of the data line. It showed me where to expect the RTTY text on the tiny display. But your plans sound good! Keep it up!

73 Harri DF5LI