df8oe/UHSDR #1708: Bug D2.11.18 and D2.11.19 in mode CW

ok1if, 2019-01-23:
In CW mode (mcHF 512kB), the TCVR does not transition into the receive mode (RX) when paddleing. It still remains in mode TX. The last OK firmware was D2.11.17.
Milan
OK1IF

db4ple, 2019-01-23:
Thank you for the report, we'll look into this!

db4ple, 2019-01-23:
Works for me, i.e. can't reproduce.

Which CW keyer mode are you using? Straight key or one of the automatic keyer modes?
Please also post other relevant CW settings...

df8oe, 2019-01-23:
I have tested intensively and cannot reproduce. But I can state that TX LED does not lit everytime you are TXing in CW. Seems to depend of the last state LED has in RX (CW-decode...)

K4LXY, 2019-01-23:
Yes - it has always been this way, through various updates.
Tnx
Howard 

> 
> I have tested intensively and cannot reproduce. But I can state that TX LED does not lit everytime you are TXing in CW. Seems to depend of the last state LED has in RX (CW-decode...)

Howard Zehr
K4LXY

m-chichikalov, 2019-01-23:
@ok1if, @K4LXY  Hi folks, could you mind share your opinions -> [there](#1706)

ok1if, 2019-01-23:
All is clear is on video.

http://leteckaposta.cz/958884798

Here is two files. First is valid and good keying with version 17 and second is for version 19. In version 19 TCVR stay in mode TX and not switch to RX mode.

Milan

m-chichikalov, 2019-01-23:
Tnk you Milan, will take a look.

m-chichikalov, 2019-01-23:
> Tnk you Milan, will take a look.

You mention that the settings the same but forgot to say what are they exactly. 
What type of key is set?
Can you put it back to RX mode by double pressing Tune button or PTT?

ok1if, 2019-01-23:
I am setting:

CW Keyer mode IAM_B and keying with paddle.

Yes double pressing Tune button rig goes to receiving mode. 

Milan

m-chichikalov, 2019-01-23:
> I am setting:
> 
> CW Keyer mode IAM_B and keying with paddle.
> 
> Yes double pressing Tune button rig goes to receiving mode.
> 
> Milan

Thank you, A bit later today I will try to understand what kind of the bug I introduced with new features ;) Thank you for the report.

ok1if, 2019-01-23:
OK, I wish you good luck.

df8oe, 2019-01-23:
Now I was able to reproduce. All works 100% ok if you come from another mode to CW. But if you start with CW the issue is present.

m-chichikalov, 2019-01-23:
I know why. Will fix shortly as get back home.

WD8BXS, 2019-01-23:
Same issue here, ALSO, it will not transmit the memories!! It goes into transmit, but does not send any letters.

TNX, de Chuck WD8BX

m-chichikalov, 2019-01-23:
Do you have any Macros in memories? Can you transmit them in RTTY or BPSK?

@df8oe I found that I'm not able to save Macros, they are gone each reboot... Is it a bug on my side or it's not implemented?

db4ple, 2019-01-23:
At least on 2.11.13 saving and restoring a macro works.

m-chichikalov, 2019-01-23:
> At least on 2.11.13 saving and restoring a macro works. (OVI40)
> 
> @m-chichikalov: You do have an I2C EEPROM installeld in your OVI40? If not, nothing is saved.

Sure, over settings stored properly as far as I know. I will try 2.11.13, thanks

PR #1711 (2019-01-24): Fix for #1708
Fixed #1708, I believe. 
UiDriver_SetDemodMode() was not called if active CW during booting.
closed 2019-01-24 by df8oe