df8oe/UHSDR #1709: Touchscreen enabled when is not present

ok1cdj, 2019-01-23:
Your firmware version: 2.11.19
Your bootloader version: 5.0.4

### Hardware 
 * UI Board: mcHF 0.6
 * RF Board: mcHF 0.6

Upgraded to latest devel version. I have no touchscreen installed in my mcHF. When mcHF boots
several times with USB keyboard then i get warning on boot screen that touchscreen is not calibrated.
I see touchscreen controller in menu.
This issue gone after several reboots without USB keyboard connected.

m-chichikalov, 2019-01-23:
Hi Андрей, have you used keyboard before? Just want to understand is it a new bug or some legacy one.

ok1cdj, 2019-01-23:
Privet Maksim, no,  never used keyboard on older versions.

73 Ondra OK1CDJ

m-chichikalov, 2019-01-23:
Thx 73!

OK2PLL, 2019-04-24:
Mám TRX mcHF ver.0.5 s modifikacemi a také modifikací na dotykový displej. Dříve bez problémů, nyní mám firmware D2.11.50 a i když při pokusu o kalibraci displeje po stisknutí displej reaguje, zobrazují se souřadnice i čísla tlačítek, tak se kalibrace neuloží a při  dalším spuštění opět požaduje kalibraci. V čem je chyba?

db4ple, 2019-04-24:
Hi OK2PLL, please post your issue next time in english AND in a different issue. Thank you.
You have to touch the screen before turning on the power on and wait until the calibratin screen appears.
You must not press any other button, only the touchscreen. Did you try this?

db4ple, 2019-04-24:
@ok1cdj 

Interesting. You are running the latest version? 2.11.19 is not the latest version but you mention having the latest version which is 2.11.54 at the moment?

OK2PLL, 2019-04-24:
Thanks, the problem has been solved. But I have another problem - sometimes 
the TRX, even though everything looks fine, is silent and there is no sound.
What can be the problem? Lev OK2PLL
---------- Původní e-mail ----------

Komu: df8oe/UHSDR <UHSDR@noreply.github.com>
Datum: 24. 4. 2019 14:41:00
"

db4ple, 2019-04-24:
@OK2PLL , please update to the latest version of Bootloader (5.0.4) and Firmware (2.11.54) and see if the issue still exists. We fixed a couple of issues which led to silent radio. If you already have the newest firmware and bootloader, then open another issue for this problem. I will not discuss this new problem here in an completely different context.

OK2PLL, 2019-04-24:
I have installed the last bootloder, as well as the firmware, unfortunately 
the problem still persists.
---------- Původní e-mail ----------

Datum: 24. 4. 2019 16:04:20
"

df8oe, 2019-04-25:
As Danilo told: Please open an new issue and describe the issue as best as you can using the issue template otherwise WE WILL NOT ANSWER OR REACT ANYWAY. It is bad behaviour to hijack threads and continue with different pointed contents!

closed 2019-08-11 by df8oe