df8oe/UHSDR #1715: sensor!

g7jur, 2019-01-26:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:
Your bootloader version: 
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

g7jur, 2019-01-26:
I have a MCHF V0.6.2. It would be nice that it did not display Sensor!, not that the chip is no longer fitted.

db4ple, 2019-01-26:
See my comment here: https://www.amateurfunk-sulingen.de/forum/index.php?board=15;action=display;threadid=1295;start=15 . Unfortunately is the english translation so bad, that even I cannot match it to what I wrote.
In a nutshell: We have no way to detect if the missing sensor is because it is a 0.6.x board or because it is not connected so we keep the message.

For the benefit of the majority of the users who don't have a mchf 0.6.x I will not submit a change which disables this message if the TCXO is switched off as it indicates a problem in any case.  We may in future have a setting to indicate that the board is of a specific sub variety and then we may turn off the message, but only when we have an idea what to do with this space. 

Honestly, it is a cosmetic problem caused by Chris', M0NKA, decision to remove the chip. We already disabled the boot delay if the TCXO is switched off just because of this change. This really was an annoyance, which we could fix without harming other non-0.6.x users too much but especially in this  case the SENSOR! message is important for them.

g7jur, 2019-01-27:
Thanks for the reply. Not sure why Chris has left off the sensor chip, or why he has gone back running the display in SPI mode

closed 2019-01-27 by g7jur

db4ple, 2019-01-27:
Yes, you have to ask Chris to get these answers. For the display I suspect that the HY28B he can get only work in SPI mode.

KS4L, 2019-03-24:
I have just discovered this as a result of building 0.6.3. Could this be reconsidered. I don't know why it is unacceptable to disable the error message if the TCXO is OFF.  It could mask an actual error with the sensor, but that would not cause a problem if the user has turned it off, and the error would display if the user attempted to turn on the sensor.  Hopefully this is a simple fix. 
73,
Randy, KS4L

kf4mot, 2019-09-02:
Just built a 0.6.3 myself and was confused by the sensor error even though I have tcxo turned off.