df8oe/UHSDR #1716: CW led setting

WD8BXS, 2019-01-27:
Your firmware version: 2.11.22
Your bootloader version: 5.0.4

 version 6 mchf
### Describe the issue:

The setting for CW LED on/off will not save.
Always reverts to ON after power down.

de Chuck WD8BXS

db4ple, 2019-01-27:
The setting CW LED on/off is not stored permanently, it is forgotten once power is turned off. However, turning off the decoder (which is stored permanently) will also disable the LED blinking (the blink control is part of the CW decoder).

WD8BXS, 2019-01-27:
OK, I understand now.  It is a pity though you can't have one without the other.

Chuck

closed 2019-01-27 by WD8BXS

db4ple, 2019-01-27:
Well, the decoder is controlling the led. It is possible to make the "CW LED OFF" setting permanently stored, but I think, the reasoning for not doing it was the need to see it to have a feedback control for the right level for the CW decoder and this level is not a fixed setting.