df8oe/UHSDR #1720: selecting mic compression to "off" breaks PA power settings

mm0mhp, 2019-01-28:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Your firmware version: D2.11.12
Your bootloader version: 5.0.2
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.4 
 * RF Board: mcHF 0.4 

### Describe the issue: ON first boot the PA power setting work as expected, Full,5W,2W,1W,0.5W all OK, however if I now select MIC Compression to "OFF" it breaks the PA power settings and TX power is always full power, if I set mic compression to anything other than "OFF" the PA power settings works again. 

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.): 
Boot radio, check PA power levels as expected, set mic comp to "off" this breaks the PA power settings and radio always delivers full power.

 
Other settings you consider as relevant: none

 * ...

mm0mhp, 2019-01-28:
My PA problem is only evident on SSB, CW is OK, when compression is "off" the SSB peaks are full power (measured on LP-100)  irrespective of PA power setting, this may be a feature and not a bug?

db4ple, 2019-01-29:
This probably indeed a feature. The compression keeps your audio level (which goes as input into the SSB modulator) within certain limits by amplifying/attenuating it so that the normal voice gets out with roughly the selected power. If you turn the compressor off, you are in control of the voice levels and thus you can overdrive it if your mic gain is high enough or you speak too  close / too loud into you mic.
You can control this by switch to show audio levels (press long on "SNAP") until you see AUD. If you keep your voice roughly below the red area, you'll sent out more or less nominal power. 
Now turn on the compressor and scream into the mic, you'll see it will try to keep the audio level in the white area mostly. Turn if off, and scream, and all gets red. Remember to have a dummy load, or you may get into trouble for screaming out loud with a lot of splatter ;-)

Once we're sure this is all like this, close this issue.

mm0mhp, 2019-01-29:
Thank you - yes you are correct, the issue is one of over modulation when compression is switched off.

closed 2019-01-29 by mm0mhp