df8oe/UHSDR #1723: FW 2.11.26 Tune crashes TRX

db4ple, 2019-01-31:
Your firmware version: 2.11.26
Your bootloader version: ALL

### Hardware 
 * UI Board: ALL
 * RF Board: ALL

### Describe the issue:

Pressing Tune crashes the TRX, need to remove power.

m-chichikalov, 2019-01-31:
Doesn't matter which mode?

db4ple, 2019-01-31:
I don't know yet. And as I have multiple independent reports for that problem, I'll guess it is easy to reproduce.

g7jur, 2019-01-31:
Same fault here, MCHF V0.6.2

db4ple, 2019-01-31:
@g7jur @ES8AS  If anyone of you could test if this crash happens in all modes (CW,SSB,FM,AM) it might help me to fix it quicker. Of course, I can test it by myself but it would save me 10-15min maybe, and that is 10-15min closer to the fix.

THX, Danilo

ES8AS, 2019-01-31:
The bug is true for SSB, BPSK and AM modes.

db4ple, 2019-01-31:
Thanks, Not for FM? Interesting...

df8oe, 2019-01-31:
FM and CW are working, the rest not.

df8oe, 2019-01-31:
Interesting. For me FM is working.

ok1if, 2019-01-31:
In CW mode is all OK. Bug only with SSB.

db4ple, 2019-01-31:
I get even more interested now after it is not happening all the time...

df8oe, 2019-01-31:
CW and FM is working absolutely stable for me (100% working). SSB , AM and digimodes are absolutely instable (100% not working).

db4ple, 2019-01-31:
@ES8AS : you will not have much fun with 2.11.25, use 2.11.24 or wait for 2.11.27

db4ple, 2019-02-02:
Fixed

closed 2019-02-02 by db4ple