df8oe/UHSDR #1726: Spectrograph bug 2.11.27

ok1if, 2019-02-01:
Look video
http://leteckaposta.cz/884407055

In version 2.11.27 spectrograph not working OK and showing false signals.

Milan OK1IF

df8oe, 2019-02-01:
Hi,

here no problems at all...

mmarzial, 2019-02-01:
Same issue ... in version 2.11.27 and 2.11.26
version 2.11.25 is OK
73 marco

[image]

[image]

db4ple, 2019-02-01:
@mmarzial : If you switch to Magnify 1x (no zoom in spectrum, does the issue disappear?

db4ple, 2019-02-01:
Problem identified and fixed. Thanks for reporting the issue.

mmarzial, 2019-02-01:
2.11.28 works 
Thank you

ok1if, 2019-02-02:
Now is OK. Thank you
Milan

closed 2019-02-02 by ok1if