df8oe/UHSDR #1728: SAM Selection

g7jur, 2019-02-01:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:2.11.28
Your bootloader version: 5.04.4
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.6.2
 * RF Board: mcHF 0.6.2

### Describe the issue:

When pressing the mode button to select AM then long press to SAM then SAM-l,SAM-U, then SAM, but it stays on SAM-U.

db4ple, 2019-02-02:
Hi, cannot reproduce this here. Does it work with an older firmware (e.g. 2.11.20) ?

df8oe, 2019-02-02:
I cannot reproduce, too. Please describe your working conditions relating to: magnify setting, used filters, DSP setting and waterfall/spectrum setting. Can you attach a screenshot of the LCD? There most relevant settings are visible...

g7jur, 2019-02-02:
Here is a screen shot

db4ple, 2019-02-02:
Screenshot not in the comment. I don't think sending an embedded image in a mail works here.

g7jur, 2019-02-02:
I will have go later. Just got a load of snow here, must go out and clear it
Philip G7jur.
Just tried it on MCHF V0.7 still the same. When you cycle though the SAM mode, and end up back at SAM the display is still at SAM-U.

db4ple, 2019-02-02:
Now I got it: It is not that cycling through the SAM modes does not work but the visualization of SAM filter bandwidth shows only the SAM-U part. BTW, it is not stuck on this, it is like this from the beginning (when you go from AM to SAM).

g7jur, 2019-02-02:
You got it. But it’s not like it from the beginning, only at the end.

sp9bsl, 2019-02-02:
Hi,
I can confirm that, long press the mode selection, and after SAM-U filter width doesn't change to SAM, twice more  and it appears as should. Bug around the bandwidh calc routine.

73 Slawek

df8oe, 2019-04-08:
Please close this issue if it is gone for you.

closed 2019-04-08 by g7jur