df8oe/UHSDR #1730: AGC allways "60" after Switch-On

DL8EBD, 2019-02-03:
Your firmware version: 2.11.29
Your bootloader version: 5.0.4

### Hardware 
SParrow H7 CPU

Issue:
after each switch-on the AGC is set to "60" again, no matter what was previously set!
No matter if SSB, CW or SAM

73!
Thomas

df8oe, 2019-02-03:
Thats correct. But I do not know if storing is useful. You have to change the value due to band, time of day etc. so that a stored value matches as rarely or as often as "60" ;) I have not looked into the code so that I am not sure it is a bug or a feature...

DL8EBD, 2019-02-03:
that's right Andreas, but "60" is a little far from the normal setting.
In my setup, the AGC is regular at 10-20. 

73
Thomas

db4ple, 2019-02-03:
AGC is normally saved. I'll check what I broke. I have the same issue but did not notice. Thanks for reporting.

db4ple, 2019-02-04:
Should be fixed now.

DL8EBD, 2019-02-04:
thanks Danilo and Andreas for the quick fix.
I close it later