df8oe/UHSDR #1731: No TX audio out on USB port

g7jur, 2019-02-03:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:2.11.29
Your bootloader version: 5.0.4
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.6.2
 * RF Board: mcHF 0.6.2

### Describe the issue:

No TX audio out on USB port

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: USB
 * Filter:  2K7 BPF
 * Noise blanker setting: 0
 * Noise reduction: OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

db4ple, 2019-02-03:
Hi,
you know that you have to activate USB audio out for TX data in the debug menu?
If not, try this. It does work for me.

g7jur, 2019-02-03:
Hi

Just tried to activate USB audio out for TX data in the debug menu to
source, still not working.
But no problems with older version of firmware.

db4ple, 2019-02-03:
Ok, then it must be an issue with the mcHF configuration of the firmware, since I was using an OVI40 for testing. 
No problem, will look into this, should not be difficult. Issue confirmed. 
Thanks for reporting.

g7jur, 2019-02-03:
Hi.
Same problem on my MCHF V0.7 as well.
Again older firmware works fine.

Regards Philip.

db4ple, 2019-02-03:
Is this in all TX out modes, or a specific one does not work (Source, Filtered, ... )? Code looks unsuspicious and is identical to OVI40.
Will have to build an mcHF firmware and test. But a hint regarding if all or only a specific TX out mode are affected, would be helpful.

g7jur, 2019-02-03:
> Yes 2.11.28 and older. Happen with the latest 2.11.29.

Yes all modes as far as I can tell.

db4ple, 2019-02-03:
I meant all tx out modes (not all radio modes like SSB, FM, AM, ... ). Not sure if I was clear enough.

BTW, I cannot confirm that there is a general issue. There is an issue with the TX OUT mode SOURCE, but the other modes (Filtered, Generated IQ, Final IQ) should work just fine (and do here).

g7jur, 2019-02-03:
Don't know then. Wont work on my MCHF V0.6.2 or V0.7., but no problems with
previous versions of firmware.
I will test it on Windows, as I am testing on a Rasberry-Pi. that could be
the reason.
Just notice that the keyboard beep does not work on the band buttons.
Regards Philip.

db4ple, 2019-02-04:
Could it be the case that you have a completely different issue from what I thought your issue report implied: Are you trying to send via for instance WSJTX from your PC to the mcHF and this stopped working on 2.11.29? 
In this case, I can tell you, I just fixed that. 

And found out that your reponses to my questions (or my questions) were not helpful to figure out what your issue actually was. I should have asked you earlier what exactly I have to do in order to reproduce the issue.  Lesson learned (well, not really, since I knew that rule before).

g7jur, 2019-02-04:
Hi.
Many thanks for your help. Sorry I am not very good at describing things in
writing.
Yes WSJT-X running on a PC, and connected to the MCHF via the USB port.
If you have fixed it then that is good new.
Regards Philip.

g7jur, 2019-02-04:
Good new just installed V2.11.31. Now working perfectly.
Many thanks for fixing so quickly.

Regards Philip G7JUR

ES8AS, 2019-02-05:
Not saving the mode of TX audio out on USB port in debug menu (source, filtered etc). After TRX switch on the mode again is OFF. (FW 2.11.32 and BL 5.0.4)

db4ple, 2019-02-05:
TX Out mode in debug menu not saved afer power-off: That is intended. We see this as debug feature which should not be enabled by accident (or default) since it adds not much value for most users.

K4LXY, 2019-02-05:
Entered an issue on Commit page - DSP NR setting not being retained when rig is shut down.  RS-918, cw mode.
Tx

> 
> TX Out mode in debug menu not saved afer power-off: That is intended. We see this as debug feature which should not be enabled by accident (or default) since it adds not much value for most users.

Howard Zehr
K4LXY

K4LXY, 2019-02-05:
Sorry:  v 2.11.32

> 
> Entered an issue on Commit page - DSP NR setting not being retained when rig is shut down.  RS-918, cw mode.
> Tx
> 
> 
>> 
>> TX Out mode in debug menu not saved afer power-off: That is intended. We see this as debug feature which should not be enabled by accident (or default) since it adds not much value for most users.
>> 
>> 
> 
> Howard Zehr
> K4LXY

Howard Zehr
K4LXY

df8oe, 2019-02-14:
I think the original issue is fixed now...

closed 2019-02-14 by df8oe

g7jur, 2019-02-14:
Yes it is.
G7JUR