df8oe/UHSDR #1742: CW memory not working

S52CQ, 2019-02-10:
Your firmware version: D2.11.25
Your bootloader version: 5.0.2
### Hardware 
 * UI Board: mcHF 6
 * RF Board: mcHF 6

I used to use CW memory quite some times, but not for last several months. Today I have tried it and it does not work.
Written messages were correctly saved, but there is no keying. Also if I record a new msg. it's correctly saved, but not played back. When I press button of the msg. to be played, radio goes to TX and remains in TX without any keying. I have to take the power, because Power button does not work.
No other issue here. I love using my mcHF.

73 de S52CQ, Jure

m-chichikalov, 2019-02-10:
Hi, could you mind to update FW for the last one and if it would not help provide more info please, what kind of settings, so on...
73.

S52CQ, 2019-02-14:
Hi,
tonight latest FW (D2.11.36) was installed, still same: CW memory is not working. It freezes radio. 
There is no other settings to be reported: enter menu for CW recorder messages, type button to play recorder msg. but radio freezes, no msg. is played (keyed). It's same if I record new msg.: recorded msg is recognised (i.e. if I key CQ radio would record CQ msg.), but radio does not reply that msg.

73 de S52CQ, Jure

df8oe, 2019-02-15:
Hi Jure,

can you please test which was the latest version that was working for you? That helps us to locate the issue much faster..

vy 73
Andreas

S52CQ, 2019-02-15:
Hi Andreas,

sure, will do that.

73 de S52CQ, Jure

m-chichikalov, 2019-02-15:
> Hi,
> tonight latest FW (D2.11.36) was installed, still same: CW memory is not working. It freezes radio.
> There is no other settings to be reported: enter menu for CW recorder messages, type button to [...]

Hi Jure, cannot reproduce. Let's assume that in F1 bank I and you have "DE RV9YW TEST". On the screen, I would see only "DE" under the button. If you turn on CW mode, press F1 button, what do you see on the screen? It should appears ">" first and after each character from message one by one after it's transmitted. Do you see only ">"?

As long as I can not reproduce, (probably others also) only you can help us to understand where the problem lies. So, any additional information welcome. 
73! RV9YW Max

S52CQ, 2019-02-15:
Hi Max,

this is trange that you cannot reproduce the error - but I realize that 
radio SW is very complex and we might have one setting different.

> If you turn on CW mode, press F1 button, what do you see on the 
> screen? It should appears ">" first and after each character from 
> message one by one after it's transmitted. Do you see only ">"?

I see only ">" - and nothing happens: radio is in TX mode, it would go 
to RX mode if I start keying with CQ key. But if I do not touch CQ key 
it keeps staying in TX mode. I can record new msg, like my call: S52CQ, 
I see it written above the button, but if I press that button nothing 
happens: radio goes to TX mode and it stays there.

I will check latest working FW and will let you know soon. I am busy 
tonight, but probably during this weekend I will check that.

73s de S52CQ, Jure

S52CQ, 2019-02-15:
Hi,

I have programmed versions starting with D2.9.84, then 2.9.92, 2.11.7, 2.11.30, 2.11.35 and the latest - and to my embarrassment (and also pleasure) - CW memory works well.

I cannot reproduce it myself, although prior to that FW uploading CW memory was not working.

So we've got the solution for future users that would have same problem as I had: upload few old FWs and CW memory will start working :)

I will wait for any comment from you guys and after few days will close this issue.

73s de S52CQ, Jure

df8oe, 2019-02-16:
I could not ever reproduce it here. Not with OVI40-F7, OVI40-H7 or mcHF... The only idea I have (but not tested yet): What if you start with a clean config where never CW memories are used AND a recent firmware? Possible recent firmware needs that you have used CW memories in the past. I have used them for testing purposes on all machines and maybe that is the reason it was always working for me. You wrote you have not used it since months... Maybe there is some initialization missing which is no longer present in newest firmwares.

S52CQ, 2019-02-17:
I suspect what was the reason:

I used to use M0NKA Bootloader and upgraded it. After that I have 
upgraded also FW (to test if FW ugrade through USB works - which of 
course it did). I think this may be the reason.

I would not worry much - now it all works nicely, thanks.

73s de S52CQ, Jure

df8oe, 2019-02-17:
OK - please close the issue for all is working now.

m-chichikalov, 2019-02-17:
> I suspect what was the reason: I used to use M0NKA Bootloader and upgraded it. After that I have upgraded also FW (to test if FW ugrade through USB works - which of course it did). I think this may be the reason. I would not worry much - now it all works nicely, thanks. 73s de S52CQ, Jure

Thank you for sharing. 73!

closed 2019-02-17 by S52CQ