df8oe/UHSDR #1743: Supported Hardware web page promotes misunderstanding of copyright as applied to hardware

BrucePerens, 2019-02-11:
### Describe the issue:
This is an issue with the web site at https://github.com/df8oe/UHSDR/wiki/Supported-Hardware

Thank you for introducing me to your project at Orlando Hamcation today. I was going through your web site and noticed a pretty severe misunderstanding regarding copyrights as applied to hardware.

Schematics and hardware designs can not, in general, be copyrighted. New inventions that are in the form of electronic designs can instead be protected by patent. But they must actually be inventions, not just another iteration of the superhetrodyne design. I will provide the U.S. law below as an example of why, law regarding copyright in other nations generally follows that of the U.S. due to treaties, lobbying, etc.

So, where the page says:

_most designers DO OWN THE COPYRIGHT TO THE PCB DESIGN AND/OR SCHEMATIC. Please respect the rules. Selling clones without an agreement with the copyright holder means breaking copyrights._

this is the way we might like things to be, but it isn't the way things are under the law. And we might wish the designers to be rewarded with sales rather than the cloners, and we can encourage people to do that. But we can't protect the designers with copyright law.

Similarly, where it says "Do not buy these, they are not licensed", it should say "Do not buy these, the creators of the design are not rewarded with any part of your purchase."

It is also not clear that all cloners are even unethical, if they are violating no law. If they do obnoxious things like represent themselves as the original designers or their product as the original, try to block sales of the original design in their nation, or send those with support requests for the clone to the original design team, those things are definitely unethical. And that is the difference between a clone and a counterfeit. I have seen these all done.

You are thus encouraged to protect your product name with trademark, as a granted trademark can be enforced should others try to use your name to misrepresent their product as yours. Unfortunately, this includes some complexity and expense and must be repeated for many nations, and actually enforcing in court is expensive too.

So, here's the second paragraph of U.S. copyright law, which rules out of copyright certain things that are supposed to be protected by patent instead:

_17 USC 102(b) In no case does copyright protection for an original work of authorship extend to any  **idea, procedure, process, system, method of operation, concept, principle, or discovery,** regardless of the form in which it is described, explained, illustrated, or embodied in such work._

We generally refer to these as _functional_ things, and functional things are the domain of patent while copyright applies to artistic creation. So, one might ask what part of software is copyrightable? Not the whole thing! Only the part where the programmer has made an artistic decision by choosing one of a number of possibilities. We think of that sort of work as _expressive_. If you are constrained by physics or some external factor to do things only one way, that part of the program is not copyrightable. Nor is work authored by a machine copyrightable, because no creativity was exercised.

So, in general we think of schematics as operating within the constraints of the world, and thus entirely functional rather than expressive. Patent can be applied to them rather than copyright.

So, what part of a schematic in a book is copyrightable under U.S. law (and many other nations). The explanation. Similarly in a cookbook it is not the recipe that can be copyrighted, but the photos and explanations around it. Chemical combinations can be _patented_. Patent expire, while copyright lives longer than people do.

Think of the problems we would have if this were otherwise: we would not be able to use the schematics that we learned in books in our products, and the practice of electronics would be stifled. Designs published in ham radio magazines back to the 1910's would have to be licensed. Even antenna designs! Even _wire_ antenna designs! Everything we did with electronics would grind to a halt. Electronics already has enough bad intellectual property law under the patent regime, we don't want more based on copyright.

One of the implications of this is that Open Hardware licenses are mostly useless without a patent. This is why I stopped working on them.

So, please fix the web site to encourage people to buy the original designs, without promoting misunderstanding of copyright.

df8oe, 2019-02-11:
Hi Bruce,
please take a look at the edited version.

vy 73
Andreas DF8OE

sp9bsl, 2019-02-11:
Hi Bruce,
you touched interesting problem, may I ask about PCB design? Is it copyrightable?

vy73
Slawek

df8oe, 2019-02-11:
Hi Slawek,
as far as I know it is NOT COPYRIGHTABLE. So the best way is to use multilayer with signal tracks in the middle layers. You can make it as difficult as possible to clone it - that's the only way. If it is easier to route a new PCB than to try to clone the existing one that is "the winner" :)

vy 73
Andreas

BrucePerens, 2019-02-13:
Hi Slawek,

In the US we have a separate form of intellectual property called the "mask work" that could conceivably apply, but I don't know of cases in which it's actually been attempted to apply it to a PCB design. Note that the work of an autorouter does not include human creativity, it is the parts of the layout that you alter by yourself that we might, weakly, enforce as protected work.

BrucePerens, 2019-02-13:
Hi Andreas,

There is still mention of "it is not licensed" on the page, which I would suggest you change to "the creators of the design are not compensated for their work by this vendor" or something similar.

Of course we wish that all nice people who share their work were rewarded appropriately. It's just that making copyright law work to do so would hurt us even more than the cloners.

    Thanks very much

    Bruce

df8oe, 2019-02-14:
I am not sure if I am willing to do that. There exist different license approaches (e.g. Creative Commons Hardware License, which Chris does use, and "Cern Open Hardware License"). I am no lawyer and I never ever have the intention to be one. But if big companies release licenses for hardware there must be in fact something where the words"license" in combination with "hardware" is appliable. So I will leave this here and I am tearing back to the license of this software piece: it is GPL v3 and that tells "the firmware and the informations here are without any warranty". So I will push the decision to everyone who gets in contact with our GitHub if he sees mcHF as "licensed" or not.

By the way: I by myself think software licenses and patents are very bad and do not help anyone (except a handful guys who get millionairs) and mostly are used to treat future and devlopment. But the world someone can imagine without such restrictions does not exist.

You say "work of a autorouter does not contain work of human creativy" - I do not confirm this. You have to interact and design many tracks by your own creativitiy because of autorouter does not know much about rf and its fancy behaviour. But I can state: Much code of commercial software does not contain human creativity, too because it is produced by computers itself. Th software license for some dexign software disallows to release products which are designed with this software as "Open Source" because the work partially is owned by the design software producer! We are running in political and very disappointing discussions here. My intention is to leave the discussion and everything around this to the ones who read the text. We do not want to destroy constructs of licenses (which of course must be prooved by a lawyer if someone wants to start mass production and worldwide selling) which are mentioned on the authors websites and respect their wishes to repeat that these clones are not licensed from the view of the authors webside.

In my view the complete commercial system which exists today is absolutely unable to help mankind to step forward and solve the actual problems of our planet and all living on it.

BrucePerens, 2019-02-18:
There are indeed problems with the intellectual property regime, etc. And what can and cannot be copyrighted, for example the work of the autorouter _without_ human modifications, could be a days-long discussion. But let's consider what your project is doing at the moment and the harm that can do:

1. You're promoting the impression that hardware designs can be protected by copyright. If courts take your example and that of other people attempting to license hardware and agree with you, we will then face a world in which most published schematics are under proprietary licenses and we will not be able to use them in our own designs. This hurts the open world.

2. You allow quite a lot of defamation of the Chinese cloners to be published on the site, and even refuse to publish information about their design as supported hardware. And yet, they are the only people in the entire world selling a fully built hardware implementation so that you can actually have a large user community for your software! They are not violating any license because there isn't a viable copyright to support the license. People allege that their component choice is inferior, but where is the proof? I bought one to examine and will make my own report. But meanwhile it seems to me that you are actively turning away users of the Chinese clone that the project should support.

db4ple, 2019-02-18:
Bruce,

I think it is fair to say that a device sold by someone should be supported by the company and dealers selling it.

Whether a non-paid hobbyist uses his or her time to support something is solely up to the hobbyist. And if a group of hobbyists decides not to support products they are not responsible for, well, that is their decision with all the implications that come from that. 

I understand your  argument of user base enlarged by users owning a "clone" and that puts us in a dilemma here.  More users of a software typically helps to make it better (if they report software issues). 

And we did accept valid issue reports from owns of commercially build devices made in China. And we fixed them. But there is no obligation whatsoever to promote these hardware products. 

Especially since some of the chinese hardware makers even violate the GPL by making changes to the software in order to use cheaper hardware parts (that is of course no problem, that is business, examples are RS-928 and RS-928plus). However, they don't publish the changes and their hardware designs. Why should we support this? It is not even "our" source running there. And I think we have the right to tell that they violate the rules of the game. 

I think we should remove the explict mentioning of hardware copyrights from the mcHF entry (I just did this). However, if a hardware project decides to publish some of their documents under a license, we should mention this. In all other cases, we point to the designer for more information.

BTW, we never actively deleted any information regarding clones from the supported hardware page (git history told me, unless I overlooked something). We surely did not encourage these vendors to use our wiki either...

K4LXY, 2019-02-19:
Like many others, I would have preferred the real thing but wasn’t in a position to build one, so had to go with the RS-918.  I’ve been very glad that you all have a been willing to support it.
Many thanks for the work you do.
73 Howard K4LXY

> Bruce,
>
> I think it is fair to say that a device sold by someone should be supported by the company and dealers selling it.
>
> Whether a non-paid hobbyist uses his or her time to support something is [...]

Howard Zehr
K4LXY

df8oe, 2019-02-19:
Hi to all,

This is my personal sight of view - I am speaking only for myself.

Not a single chinese cloner has published the schematics or the changes he did to it. No one has contacted us and told us where are specific changes so it is impossible to support them fully. They have got many informations and work for free and do not publish the small changes they made. Hmmmm...

We are "just working on the software side". World does actually not work as an "Open World" and there are many restrictions (mostly made by the "global players" of the capitalistic game). I think in this world we are acting unfair if we completely ignore the wish of enthusiastic hobby developers. If you do so you some of them will no longer publish their work. Is that a wanted solution? Not everything is for everyone! You cant get a helicopter if you do not have the money to buy one. And Chris idea was his project is something for homebrewers and not for hams who wish to buy complete devices. So he explicitely choose a license "no commercial use". It would be fair if those who cannot build this by themselves do accept this sentence and buy another rig. There are many on the market. Do you think the wish of hams who want to buy complete devices give you the philosophical right to override his wish "no commercial use?"

For example:
I am not sure you do know it is not as easy to sell kits or complete devices in Europe or USA. There are laws which restricts you (e.g. electronic waste regulations, regulations about unwanted radiation...) If a seller lives in Europe or USA these laws will it make much more dificult to sell this legally and will make the price much higher. If you are located in China you can completely ignore these laws and produce much cheaper (and at quite different standards). The devices are imported by the buyers themself - *they* are responsible for confirmance to the laws. Do you have thought about all these arguments before you fight for the cloners?

I think you are mixing up things which are not comparable. You tell "there is no possibility to copyright schematics". Nobody wants to copyright these thing FOR HOMEBREWERS / for everyone! You are telling "...if these things are copyrighted..." ...nobody can use it. Correct. But noone wants to copyright it in a way that NOBODY can use it. In case of mcHF the only wish of the author was and is "no commercial use". So you can use his work in any way you want - but (as he says) do not make a business from it. I think to put your wish to get complete devices over the wish of the author and accept the only way to become your wish true is to ignore him is thing like egoistic and I cannot see any ham spirit here...

Again: this is *my personal view of the things*.

And because of I know what is going on outside in the world I am designing new OVI40 RF in a way that cloners do not have easy game. Of course this do make development time much longer and the costs are little bit higher - but you need more than a scanner and software to make Gerbers from a scanned PCB. You cannot protect the work by law and you cannot vmake it completely impossible to cloners to do their business - but you can make the work as difficult as you can :)

vy 73
Andreas

DD4WH, 2019-02-19:
As has already been said: I think the keywords that are most important here are:

*HAM SPIRIT*

*HOBBY*

*SOFTWARE*  not hardware.

My personal position: if I should feel threatened to have to struggle with legal subtleties of hardware licenses here on github, I suspect my desire to contribute code here will quickly decrease to zero. I will keep the code private then.

Please do not waste your time any longer here on license discussions, invest your time into C and contribute to the code :-).

73s Frank DD4WH

df8oe, 2019-02-19:
Fully agree. The mentioned discussiond should firstly take place in the round of companies and politicians . These should firstly create a base that everyone can live without financial problems so that he can contribute to the worlds knowledge without personal loss of life standard. The "new world" (which I would strongly welcome) must grow from upper and lower (interested watching Jeremy Rifkins thoughts about that)...

closed 2019-08-11 by df8oe