df8oe/UHSDR #1749: Mode change fail

ES8AS, 2019-02-28:
Your firmware version: 2.11.37
Your bootloader version: .5.0.4
UI Board: mcHF 0.6
RF Board: mcHF 0.6

After update to the latest version I've found the bug (?). When I changed MODE to FreeDV I can't change MODE to next. The TX led is ON. The rig is stoned. I can't switch it OFF, only power plug out.

df8oe, 2019-02-28:
I can reproduce that freedv freezes on mcHF now after a few seconds (first test was made with too short working time...)

db4ple, 2019-02-28:
I obviously tested not with long enough TX. Will look into it later...

db4ple, 2019-02-28:
I can not reproduce. Please provide configuration backup via uhsdr_tool so that I can test with your configuration data.

df8oe, 2019-03-01:
Toggle through the modes using MODE button. When you reach DI-L (or DI-U) wait a few seconds. McHF will freeze completely showing red flashing LED.

[settings.zip](https://github.com/df8oe/UHSDR/files/2917997/settings.zip)

db4ple, 2019-03-01:
Thanks, without your settings I can toggle through the modes without any problem it seems.
I still have to  check what happens with the settings applied, though.

df8oe, 2019-03-01:
The problem is present using "default config", too...

db4ple, 2019-03-01:
We might  have an issue in the FreeDV codec2 code itself, but I am not sure yet.

df8oe, 2019-03-01:
F7 and H7 does not show the issue. So my guess is a race condition at F4 or running out of RAM...

ac0ax, 2019-03-10:
I'm not sure if this is the same problem that I'm having with 2.11.40 but when I select freedv I lose audio when I select a different mode or after a few moments. It does on both of my radios. One has 512k and the other 1M

db4ple, 2019-03-10:
Not clear what you mean: 
a) You have no audio when in FreeDV? 
b) You have no audio in different mode after having been in FreeDV?

ac0ax, 2019-03-10:
Both

df8oe, 2019-03-11:
I cannot confirm observations of @ac0ax - audio in FreeDV stays regardless how long I am listening to it. Additionally audio of SSB is present when I change modes immedialtely or after minutes.

But I can confirm that receiving of FreeDV is not working as expected on mcHF. After a few seconds of silence audio sounds scrambled. You can hear different beepings which are changing in frequency and if you tune to a (working) FreeDV signal nearly nothing is decoded. You can watch some milliseconds of decoding during a period of 10 seconds but no continuous decoding. McHF does not hang and changing to other modes results in working modes - except you reach again FreeDV.

db4ple, 2019-03-11:
Hi,
I will look into this later this week. What I would like to get from @ac0ax is the config extracted using the uhsdr_tool (See Wiki Backup and Restore). You can attach these to a comment by drag and drop.

martinling, 2019-03-11:
I'm seeing the same as @ac0ax, running commit `ae70cc99` on mcHF 0.6 with 2MB flash and 256KB RAM. On switching to the FreeDV mode (with no actual signal being received), I get periodic short bursts of garbage audio, and then when switching back to SSB/CW etc I have no audio at all and have to reboot to get it back.

df8oe, 2019-03-12:
Please provide your configuration file. Not reproduceable here. I get garbaged audio when listening to freedv but if I switch back to SSB all is ok again. There are so many possible settings that can impact this that we need the complete config.

df8oe, 2019-03-12:
I have reproduced "by chance". If NR is enabled I can reproduce it. But I very, very rare activate NR ;)

db4ple, 2019-03-12:
Hi,
thanks Andreas. Some here, since I mostly use digital modes, NR is most often off. Will look into this later as said earlier.

martinling, 2019-03-12:
@df8oe pretty sure I had this happen with NR off as I rarely use that either. I can extract a config when I'm back at the radio.

df8oe, 2019-03-12:
@martinling that would be helpful. I cannot reproduce with DSP OFF.

martinling, 2019-03-12:
@df8oe here's my config. Just tested again and definitely happening with DSP off.

[uhsdr_config.zip](https://github.com/df8oe/UHSDR/files/2958942/uhsdr_config.zip)

db4ple, 2019-03-17:
Can you still see the issue? If not, please close!

closed 2019-03-17 by ES8AS