df8oe/UHSDR #1750: Tune changes output level

K4LXY, 2019-03-02:
Your firmware version: 2.11.36
Your bootloader version: 5.0.4
UI Board: mcHF 0.6
RF Board: mcHF 0.6

After using TUNE, the power out level changes to that set for tune.
For example, I am using FULL power.  I push TUNE, which is set at 1 W.
When tuning is done, the output level is now set at 1 W instead of FULL.
I didn't notice at what firmware update this change happened.  
Not a big problem, however.
Tnx
Howard K4LXY

 * ...

K4LXY, 2019-03-03:
Thanks! Now power level returns to set value after tune.
73 Howard

K4LXY, 2019-03-15:
* UI Board: mcHF 0.6
 * RF Board: mcHF 0.6
Firmware 2.11.42
Bootloader 5.0.4

This is basically a followup on the issue I raised earlier, then closed.  In cw, if the power is set to, say, 5W, and the Tune level set to .5w, when I push Tune the power indicator says .5, but it actually puts out 5W.  This will only be seen if actual output monitor is monitored.

Again, certainly not a major problem.
73
Howard K4LXY

ase-2017, 2019-03-15:
Hi,
I can confirm the issue with 2.11.40. Will look into it.

K4LXY, 2019-03-15:
I'm measuring output with an external wattmeter (several, actually, and they all agree).
Howard

db4ple, 2019-03-15:
Hi Howard,
can you test 2.11.45 and report.
Thank you!

K4LXY, 2019-03-15:
Just tested.  When power is set at FULL and tune at .5, I get 1.5W out.  On the lower power settings (e.g. 5W, 2W), I get .5 out as it should.  I'd say it's working.   Thanks for working on this!
Howard

df8oe, 2019-03-15:
I do actually have access to 4 mcHFs. No one of this tunes with anything except "as TX power". Very strange. Additionally none of them shows anything on power meter - regardless of config "swap fwd/rev".

db4ple, 2019-03-15:
I'd say these may be very loosely related issues. 
Howard basically confirmed my measurements that 2.11.45 does the right thing.

And the issue in the code I found was clearly linked to the symptom we saw and also explains why your fix worked. In fact what was necessary to set the correct power were two consecutive calls to the RadioManagement_SetPowerLevel function in order to set the correct power factor (the first call calculated a new internal power value, unfortunately the function for the power factor used the stored ts.power value, not the new one, then we stored the newly calculated power value in ts.power , then the second call finally used this one and voila, we have the right power.

My changed simply put things in right order. It did not change anything else.

@df8oe: Is the power meter showing output power when tuning using "as TX power" or is it generally "dead"?

db4ple, 2019-03-15:
@K4LXY: You're right, with normal tx power set to full power we were using the wrong power factor. I'll fixed that, but we still have to figure out what is the problem for @df8oe:
- if the PA calibration for FULL is not done, tuning with full power set as normal tx power will cause no power whatsoever being emitted because of the just found issue (so full power calibration not done AND tune power set to anything but FULL -> zero tune power in 2.11.45). In 2.11.44 this would not have caused issues, because calling the function twice would mask this issue, too.

K4LXY, 2019-03-15:
PA calibration for Full has been calibrated for 10W.

Howard

df8oe, 2019-03-16:
Using 2.11.46 tune power level is set correctly. Flashing back to 2.11.45 reinvolves the issue. So tune power level now is working as expected. But SWR and Po meter both do not show anything. This is already for a longer time and has nothing to do with tune power level. I do not have mentioned it because I use external SWR measure.

db4ple, 2019-03-16:
@df8oe : Can you confirm that Power Factor for Full power was not set (or set to very low values) on the mcHFs you tested?

@df8oe: Shouldn't we open an issue, if there is no power meter and swr meter output on mcHF to see if there is someone else with this issue? On the OVI40 I have PO meter & SWR.
Unfortunately, my my second mcHF PA which I use with the mcHF UI does not have the finals fitted, so I cannot test this easily.

closed 2019-03-16 by K4LXY

df8oe, 2019-03-16:
Full power setting is set to 10W on all bands. "Reduce Power on xx Bands" is set both to OFF and full power values are between 20...55.