df8oe/UHSDR #1754: CW memory 2nd time

S52CQ, 2019-03-10:
Your firmware version: D2.11.36
Your bootloader version: 5.0.2
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 6
 * RF Board: mcHF 6 

Hi,

CW memory kept working and not working so I have decided to find step-by-step setting that replicates the error.

1. Turn ON mcHF
2. Set VFO B. mode: USB, 7.074 MHz, set VFO A to 7.030, mode CW, long press M1, record a CW message (if not previously recorded), after that select VFO B
3. Turn OFF mcHF
4. After few seconds turn ON mcHF, it will have settings like in step 2.
5. Now select VFO A
6. Long press buttom M1 to enter CW memory
7. Play recorded CW msg: it will not play, it will just hang in TX doing nothing. To exit that, just press CW keyer.
8. Select different mode (LSB and back to CW: now all CW memories work well.

Best 73s de S52CQ, Jure

m-chichikalov, 2019-03-11:
Thank you, I will try to reproduce a bit later on this week.

S52CQ, 2019-03-26:
Hi Maksim,

any news on that issue?

73s de S52CQ, Jure

m-chichikalov, 2019-03-26:
HI, I was really busy these last two weeks on the field tests. Going spend some time on it during this week. I let you know. 73!

df8oe, 2019-03-30:
Does the fix work for you?

S52CQ, 2019-03-30:
I did not realize it was done. Will check it within few days and will 
let you know.

73s de S52CQ

S52CQ, 2019-03-31:
Yes, It works now!

Great job guys!

73s de S52CQ

m-chichikalov, 2019-04-21:
Hi @S52CQ, Seems like you reopened it, did you find something more?

S52CQ, 2019-04-21:
Hi guys,

today I was working S5 KVP contest. CW memories worked well. There are however two things that I'd like to check with you:
After (any) memory is keyed there is quite long delay when radio stays in TX mode - but it should as quickly as possible switch to RX because in contests things are happenig quickly. Is it just my radio or settings that I have wrongly set?
I have noticed that Memory 3 always stores a letter K o the ed, o matter what I record into it. Again, is it just my radio or am I doing womethig wrong? Memory 1 ad " are keying excacly what I have recorder into them.
73s de Jure, S52CQ

S52CQ, 2019-04-22:
Yes, as I have written: when memory is keyed there is a very long delay 
- can we remove that delay after last letter has been keyed?

73s de Jure

m-chichikalov, 2019-04-25:
Hi, This is what I've measured
[image]
Will think on weekend how to improve it. Seems like it's depends on the CW speed and slightly more than pause between symbols...

S52CQ, 2019-04-25:
After last character there should be no delay, no matter of CW speed. Just delete delay after last stored character :)

S52CQ, 2019-04-25:
And "k" on the end? Where does that come from? This should be deleted too. It should store only recorded characters.

m-chichikalov, 2019-04-25:
'K' was the last recorded character. BTW, how do you record msgs? Via usb keyboard or by paddle?

S52CQ, 2019-04-25:
I did mention that "k" is somehow added to Memory 3 on it's own. 
I record msgs via paddle. I did not even know that it can be recorder via USB :)

m-chichikalov, 2019-04-25:
Yep, check it out, you can replay recorded msgs by pressing F5..F7 keys on the key-board... should be very helpful in Contest...

S52CQ, 2019-05-18:
This evening I was testing the ewest FW D2.11.55 of May 1st. It appears that in CW recordig mode Memory button 3 records well (without adding K at the end) however, when playing Memory 3 it plays memory correctly but just remains in TX. It returns to RX when I touch keyer.

S52CQ, 2019-08-19:
I have spoken with another HAM today. He suggested that this K on the end might be "left over" from previously recorded msg. that was not deleted completely to the end so the K is generated.
Is there any procedure to delete all CW memories?
73s de S52CQ, Jure

m-chichikalov, 2019-08-19:
As simple approach you can try to write another message into the long enough to overwrite old one + the mystery K.

m-chichikalov, 2019-08-19:
@S52CQ regarding delay after last character.
What value of CW RX delay set in configurations on your side?
This value in config show not real delay... 
If set this value to 
     0ms the delay will equal `one-dit + 667uS`
if 10ms -> `one-dit + 34ms`
if 20ms -> `one-dit + 67ms`

I will push PR to adjust it a bit later but could you try to set to 0 and measure delay?

db4ple, 2019-08-19:
Well, the cw tx to rx delay is a multiple of 33ms (50*0.667us)  (plus one dit if using a keyer mode), see https://github.com/df8oe/UHSDR/blob/0c7e1fc642e5bf6b419489d85414c81e01ccfa5a/mchf-eclipse/drivers/audio/cw/cw_gen.c#L412
The displayed delay time in the menu is misleading (read: wrong).

S52CQ, 2019-08-20:
Tonight I was testing Ver 2.11.64. It turns out that the annoying K at the end of the third memory is gone however, tranceiver remains in TX after generating recorded msg. It goes back to RX if I touch CW keyer.
@m-chichikalov: If I set CW TX-> RX delay to <70ms then the radio remains in TX after any recorded msg is generated, not only third memory.

m-chichikalov, 2019-08-21:
@S52CQ  tnx, will try to reproduce.

m-chichikalov, 2019-08-22:
@S52CQ  Hi, could you post all your cw settings and text in macros?

S52CQ, 2019-08-22:
Hi,
here are CW settings.
CW keyeer: IAM A
Keyer speed: 21
Keyer Weight: 1.0
Sidetone Gain: 9
Side/offset Freq: 750 Hz
Paddle reverse: Off
Tx-> Rx delay: 70 ms
Freq. >Offset: SHIFT
LSB/USB: USB
decoder enable: OFF
Signal treshold: 32000
Tune helper: ON
Blocksize for Goertzel: 88
Noise cancel: OFF
Spike Cancel: OFF
AGC for decoder: OFF
Show CW LED: ON

Macros are: 
1st. memory: S52CQ
2nd memory: R5NN
3rd memory: ?

73s de S52CQ, Jure

S52CQ, 2019-08-28:
Hi, great job! It all works well with old ad new macros.
73s de S52CQ, Jure

closed 2019-08-28 by S52CQ