df8oe/UHSDR #1755: Tune power set ignored

K4LXY, 2019-03-10:
Your firmware version: 2.11.40
Your bootloader version: 5.0.4

### Hardware 
 * UI Board: mcHF 6
 * RF Board: mcHF 6

### Describe the issue:
In cw mode (others not tested), TUNE uses whatever the power level is set at on the display, NOT the tune level set in the PA configuration.  Not a big problem, though
Howard K4LXY

db4ple, 2019-03-10:
Hi Howards,
please check your settings. 

I just tested to set PA Tune power to 0.5 W in PA Config menu and set power to Full. When I press tune, tune power display switches to 0.5 and output power is accordingly low.

K4LXY, 2019-03-10:
HI
I had the tune level set at 1 W in PA Config men, and the power set to Full.  It tuned at full power.  I tried setting power to 5W.  It tuned at 5W.
I’ll try setting to .5W in PA config and see if that makes a difference, but it will be later today.
Again, this is not an urgent problem.
Thanks

> 
> Hi Howards,
> please check your settings.
> 
> I just tested to set PA Tune power to 0.5 W in PA Config menu and set power to Full. When I press tune, tune power display switches to 0.5 and output power is accordingly low.

Howard Zehr
K4LXY

K4LXY, 2019-03-10:
I had a minute before I have to leave so I set tune level at .5W in PA Configuration.  Same thing - it tuned at the level shown in the display (5W).
I really don't mind this, and it may just be something in my configuration.

K4LXY, 2019-03-11:
If I'm the only one experiencing this, I'll close it.

closed 2019-03-11 by K4LXY