df8oe/UHSDR #1763: Pout meter and SWR meter are always dark

df8oe, 2019-03-16:
Your firmware version: 2.11.46
Your bootloader version:  5.0.4

### Hardware 
 * UI Board: mcHF 0.4  MCU STM32F439
 * RF Board: mcHF 0.4 with all modifications

### Describe the issue:
Neither Po meter nor SWR meter does show anything. All bars are dark during TX. Power out is ok. "SWAP FWD/REF" does not change anything. VSWR protection is set to OFF. Coupling factor is adjusted on all bands (in former times Po and SWR have worked flawlessly).

db4ple, 2019-03-16:
I now added finals to my mcHF RF so that I could test TX with the mcHF UI. And it works as expected (more or less). It definitely shows a power output and I verified the mcHF power display with an oscilloscope. Looks good! So I cannot reproduce the issue. 

I think, you'll need to go back and identify the firmware which broke the functionality.

df8oe, 2019-03-16:
Not a problem at all. I will test the next days and report.

db4ple, 2021-02-19:
Closed due to inactivity

closed 2021-02-19 by db4ple