df8oe/UHSDR #1772: Hello , I am wondering if New serial firmware 2.11.XX has Better Hilbert filters compared to 2.9.XX versions , Rx sound quality seems have better punch , i am wrong ?? did some thing really changed on sound filtering and quality with new serial firmware 2.11.XX  thanks a lot for clarifications 73's

nizar33, 2019-04-14:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:
Your bootloader version: 
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

nizar33, 2019-06-23:
Hello , and congratulation for excellent firmware, pse when touch screen pushed on spectrum some time we hear the Beep sound but without frequency changes , can we hear beep only if frequency will be changed to avoid confuse between valid touch screen valid pressed action and invalide ones ,  second the frequency step on touch screen to tune was fixed to 2Khz , can we reduced to 1Khz to be more confortable ....   thank you very much   73's  Nizar

df8oe, 2019-07-15:
Between 2.9 and 2.11 are universes - sorry, I cannot tell you without deep digging into hundreds of commits if there are any changes in filtering - I do not have the time to do this. But if you are interested you can look at the commits by yourself - you do see the same as I do...

vy 73
Andreas

nizar33, 2019-07-18:
Thank you very much Andreas for response
73s   Nizar

closed 2019-07-29 by df8oe