df8oe/UHSDR #1782: Request ACC out tail time configurable

RightRudder, 2019-06-21:
I am starting to use a separate RX and TX antenna setup with my mcHF.  I noticed in the config menu I can configure the initial mute time and confirm that it works on my scope, but I wish I could also have a menu item to set the tail or release time for the opto-coupled output.  It would be nice to be able to listen on the TX antenna momentarily at the end of each transmission before automatically switching over to the RX antenna.  That way I always know which antenna hears better.  It currently looks like tail time is fixed at around 60msec but it would be nice if I could configure it for longer up to perhaps 1000msec.  Could this menu item be added?

Best regards
Joe ve3vxo

df8oe, 2019-06-21:
Hi,
ACC driving is coupled in hardware by switching from TX to RX. So if we add a delay after you released PTT it is equal to pressing PTT for a longer time. I do not see any technical sense in this feature wish. Please clarify where is the advantage.

vy 73
Andreas

RightRudder, 2019-06-22:
I see now PTT_N5V is only derived from PTT_3V.  I missed that and was hoping PTT_N5V could be held after transition from tx to rx so that the receiver gets to listen on the Tx antenna for a fixed time before ACC signal drops and reciever gets the Rx antenna again.  Of course I could switch the antenna manually but I don't like to use a manual switch. Too easy to forget to switch and then transmit on the wrong antenna.  Looks like I will have no choice but to generate the timing externally.  Thank you for your comment Andreas.  73  Joe

closed 2019-06-22 by RightRudder