df8oe/UHSDR #1784: Latest commit b79484d build fails

roelandjansen, 2019-07-14:
Latest commit b79484d seems to fail the build here:

after pulling the latest code:

  [CC] drivers/ui/encoder/ui_rotary.o
  [CC] drivers/ui/radio_management.o
drivers/ui/radio_management.c:117:5: error: initializer element is not constant
     bi_80m_r2,
     ^
drivers/ui/radio_management.c:117:5: note: (near initialization for 'bandInfo_combined[0]')
drivers/ui/radio_management.c:118:5: error: initializer element is not constant
     bi_60m_gen,
     ^
drivers/ui/radio_management.c:118:5: note: (near initialization for 'bandInfo_combined[1]')
drivers/ui/radio_management.c:119:5: error: initializer element is not constant
     bi_40m_r2_3,
     ^
drivers/ui/radio_management.c:119:5: note: (near initialization for 'bandInfo_combined[2]')
drivers/ui/radio_management.c:120:5: error: initializer element is not constant
     bi_30m_all,
     ^
drivers/ui/radio_management.c:120:5: note: (near initialization for 'bandInfo_combined[3]')
drivers/ui/radio_management.c:121:5: error: initializer element is not constant
     bi_20m_all,
     ^
drivers/ui/radio_management.c:121:5: note: (near initialization for 'bandInfo_combined[4]')
drivers/ui/radio_management.c:122:5: error: initializer element is not constant
     bi_17m_all,
     ^
drivers/ui/radio_management.c:122:5: note: (near initialization for 'bandInfo_combined[5]')
drivers/ui/radio_management.c:123:5: error: initializer element is not constant
     bi_15m_all,
     ^
drivers/ui/radio_management.c:123:5: note: (near initialization for 'bandInfo_combined[6]')
drivers/ui/radio_management.c:124:5: error: initializer element is not constant
     bi_12m_all,
     ^
drivers/ui/radio_management.c:124:5: note: (near initialization for 'bandInfo_combined[7]')
drivers/ui/radio_management.c:125:5: error: initializer element is not constant
     bi_10m_all,
     ^
drivers/ui/radio_management.c:125:5: note: (near initialization for 'bandInfo_combined[8]')
drivers/ui/radio_management.c:126:5: error: initializer element is not constant
     bi_6m_r2_3,
     ^
drivers/ui/radio_management.c:126:5: note: (near initialization for 'bandInfo_combined[9]')
drivers/ui/radio_management.c:127:5: error: initializer element is not constant
     bi_4m_gen,
     ^
drivers/ui/radio_management.c:127:5: note: (near initialization for 'bandInfo_combined[10]')
drivers/ui/radio_management.c:128:5: error: initializer element is not constant
     bi_2m_r2_3,
     ^
drivers/ui/radio_management.c:128:5: note: (near initialization for 'bandInfo_combined[11]')
drivers/ui/radio_management.c:129:5: error: initializer element is not constant
     bi_70cm_r2_3,
     ^
drivers/ui/radio_management.c:129:5: note: (near initialization for 'bandInfo_combined[12]')
drivers/ui/radio_management.c:130:5: error: initializer element is not constant
     bi_23cm_all,
     ^
drivers/ui/radio_management.c:130:5: note: (near initialization for 'bandInfo_combined[13]')
drivers/ui/radio_management.c:131:5: error: initializer element is not constant
     bi_2200m_all,
     ^
drivers/ui/radio_management.c:131:5: note: (near initialization for 'bandInfo_combined[14]')
drivers/ui/radio_management.c:132:5: error: initializer element is not constant
     bi_630m_all,
     ^
drivers/ui/radio_management.c:132:5: note: (near initialization for 'bandInfo_combined[15]')
[... 204 lines cut ...]
drivers/ui/radio_management.c:196:9: error: initializer element is not constant
         bi_23cm_all,
         ^
drivers/ui/radio_management.c:196:9: note: (near initialization for 'bandInfo_region3[13]')
drivers/ui/radio_management.c:197:9: error: initializer element is not constant
         bi_2200m_all,
         ^
drivers/ui/radio_management.c:197:9: note: (near initialization for 'bandInfo_region3[14]')
drivers/ui/radio_management.c:198:9: error: initializer element is not constant
         bi_630m_all,
         ^
drivers/ui/radio_management.c:198:9: note: (near initialization for 'bandInfo_region3[15]')
drivers/ui/radio_management.c:199:9: error: initializer element is not constant
         bi_160m_all,
         ^
drivers/ui/radio_management.c:199:9: note: (near initialization for 'bandInfo_region3[16]')
drivers/ui/radio_management.c:200:9: error: initializer element is not constant
         bi_gen_all,
         ^
drivers/ui/radio_management.c:200:9: note: (near initialization for 'bandInfo_region3[17]')
drivers/ui/radio_management.c:214:28: error: initializer element is not constant
 const BandInfo *bandInfo = bandInfos[0].bands;
                            ^
make: *** [Makefile:264: drivers/ui/radio_management.o] Error 1
roeland@taplop:~/src/UHSDR/mchf-eclipse>

db4ple, 2019-07-14:
Please report which compiler version and package you are using on which operating system! Thank you!

We are running Travis Build tests for the commits (for this commit: https://travis-ci.org/db4ple/UHSDR/builds/558511560?utm_source=github_status&utm_medium=notification)  and these pass without problems. 

Most likely your compiler is "too old". Did you do other modifications to the Makefile (or code) or are you building the commit unmodified?

roelandjansen, 2019-07-14:
- no modifications on the source code. 
- I am pulling the commit -- hence it gets merged. 
- opensuse 15.1
- cc1 is in /usr/libexec/gcc/arm-none-eabi/5.2.0/ so let's guess (and I'll double check) 5.2.0.

db4ple, 2019-07-14:
OK, I verified, the issue also exists if we build with Travis and gcc 6 and 7. Turns out, only gcc 8 happily compiles this code.
As a quick fix, I recommend to switch to use the ARM gcc packages, which you can simply untar somewhere and use as per instructions in our Wiki (OPT_GCC_ARM=... make ...).

We'll look into this.

roelandjansen, 2019-07-15:
I'll see if the other projects can handle an update as well; the repo I git the compiler from has not been updated in a while (read: empty).
(md380 tooling and adalm pluto)

df8oe, 2019-07-15:
Please take the toolchain from this link: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

roelandjansen, 2019-07-15:
Andreas, 

that would be a very very very very very very very very last resort. In general  if it's not packaged, it won't be installed. ;-)

Anyways, I may have cross-arm-none-gcc8-8.3.1+r271793-1.2.x86_64.rpm and friends. I'll check  the versioning against your link and see how much it differs (it is v8 for sure).

df8oe, 2019-07-15:
Sorry Roeland - but this should *always be the first resort*! Or you do have a Distro with very recent software like Arch based distros. It took only 3 days from release of the new gcc to arrival at the "official repos" :) If you want to have newest software using other distros you normally have to leave the way of "packages"...

db4ple, 2019-07-15:
Hi,
the recommended "package" is not interfering with your normal installed software. You just unpack it in a directory of your choice, that's it. And you tell OUR makefile via OPT_GCC_ARM variable where this directory is (well, the bin directory inside this directory structure). No other project would even notice you use the compiler. That is the beauty of it. Anyways, I understand your resistance to install just another compiler, but at some point we may drop compatibility with gcc 5, as this is a fairly "old" compiler and newer versions are easily accessible for use on any distribution, see above.

However, my fix in #1785  will bring back compatibility in this particular case and it was surely a good thing you reported the problem as it was a problem which had to be fixed.

roelandjansen, 2019-07-15:
thanks. I'll let you know.

ps it's not that a different compiler is the biggest issue -- it's the fact that it's not packaged (yet).

roelandjansen, 2019-07-15:
the build went OK.

closed 2019-07-15 by roelandjansen

db4ple, 2019-07-15:
Well, this "package" is intentionally not packaged as a distribution-specific package by ARM so that it can be used exactly in this manner on different distributions and you can also have many different packages in parallel without problems. Since a cross-compiler is fairly self-contained and does not have real external dependencies, it is fairly easy to do.

Anyway, you don't have to like it or use it.