df8oe/UHSDR #1786: no PA output above 30m band (included) with >= 2.11.57

df8oe, 2019-07-15:
I can confirm there is nearly no TX output with firmwares 57/58 on bands 30m and higher. No effect changing "Reduce Power on high Bands". Seems to be a bug...

vy 73
Andreas

db4ple, 2019-07-15:
It works for 80m and 40m? What is the shown color and value of the power indicator for 30m?

df8oe, 2019-07-15:
If you mean the blue box: it stays blue and font is white - same as on 80 and 40...

db4ple, 2019-07-15:
Yes, this was what I wanted to know. Interesting, that 40m works and 30m doesn't. Do you hear the relays operating when switching bands between 60m and 40m and vice versa?

db4ple, 2019-07-15:
Please tune between bands by turning the dial and watch if the effective band indicator changes accordingly (the one with "Bnd" , to the right of the frequency) and if you can hear the relays operating when crossing the 4, 8 and 16 Mhz borders.

df8oe, 2019-07-15:
Identifed :) ! Band relays are not switching all the time. They switch if I cross 4MHz but that is the only border they are switching. Neither changing from 80m "down" to 10m leads to relay switching nor switching from 40m to 30m.

db4ple, 2019-07-15:
But all the display indicators are working as expected, right?

db4ple, 2019-07-15:
Issue identified. Look at the 1600000 and 3200000 values in   mchf_rf_bandFilters: 
https://github.com/df8oe/UHSDR/blob/88acf3c8318c1040a0f81afa4b56369ee1bcfc16/mchf-eclipse/drivers/ui/radio_management.c#L1062

Well. If it only would be 16 and 32 Mhz...

df8oe, 2019-07-15:
:) fixed now.

closed 2019-07-15 by df8oe