df8oe/UHSDR #1791: How to modify for a wide filter TX?

sv1rki, 2019-07-29:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:
Your bootloader version: 
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

df8oe, 2019-07-29:
You can find all filter coefficients here:

mchf-eclipse/drivers/audio/filters/

vy 73
Andreas

db4ple, 2019-07-29:
More specifically:

This where a tx filter is being activated:
https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/drivers/audio/tx_processor.c#L72
void TxProcessor_Set(uint8_t dmod_mode)

This is were most (read: all non-FM mode) filters are defined:
https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/drivers/audio/filters/iir_2k7_tx_bpf.c

We also have bass and treble filters, which can be used to amplify/dampen certain voice components,
see also the tx_processor.c for more information.

Please note, there is also a filtering taking place at the IQ level (during/after iq signal generation)

df8oe, 2019-08-11:
If this answers your question, please close the issue.

closed 2019-10-11 by df8oe