df8oe/UHSDR #1795: Ver 2.11.63 Band up/down issue

S52CQ, 2019-08-11:
Your firmware version: 2.11.55
Your bootloader version: 5.0.2

### Hardware 
 * UI Board: mcHF 0.6
 * RF Board: mcHF 0.6

Hi,
I have just uploaded Ver. 2.11.63 and noticed that I cannot change band from 1.8 to 28 MHz with Band Down switch. And same is from 28 MHz to 1.8 MHz with Band Up switch. It all works well on build ver. 2.11.55 that I had lastly uploaded and used.

73s de S52CQ, Jure

df8oe, 2019-08-11:
Hmmmm...... Is working for me....

df8oe, 2019-08-13:
Please leave a feedback if ver. 64 fixes your problem.

S52CQ, 2019-08-13:
Hi, yes it's better now, but when going up from 28 MHz I jump to 12,500.250 MHz and if I press again Up I go to 1.8 MHz. Same is in the Down direction I go from 1.8 to 12.5, and then to 28 MHz,

73s de S52CQ

df8oe, 2019-08-13:
I cannot confirm here on none of my four mcHF/OVI40. All is working as expected (and as before). It must be settings-related and I do not have any idea what setting is responsible... Let us wait if someone else reports this.

S52CQ, 2019-08-13:
Hi,
well, what can I tell - It's just great! This version has made my mcHF to work on 6m! Working both Rx & Tx with limited performances of course. So I will leave this version as it is as I will test mcHF for performance on 6m.
73s de Jure

db4ple, 2019-08-13:
Of course, and if you would have told us that the switch from 1.8 Mhz to 28 Mhz did work (just not to the expected 28 Mhz) we could have told you this right away. BTW, did tx at 6m will really work?  Above 32 Mhz we disable the switch to TX.

S52CQ, 2019-08-13:
Well, ver. 63 did not work - as reported, latest works with this little 
"improvement" :)

Yes, TX & RX actually worked, of course TX had very little power, but it 
was heard on the other radio on my table. Same applies for RX.

Could you enable TX switch on 52 MHz in future versions? I realise that 
RF board is not tuned for that but for experimentation it would be great 
to have 52 MHz band too.

73s de S52CQ

df8oe, 2019-08-14:
I think you have had corrupted frequency in your "6m memory place" (something like 12.5MHz...) and that confuses you....

6m TX is very problematic because of BPFs, used impedance matchin from mixer to drivers and 90° stability if you do not use potato flip-flops. So we are actually not sure if we enable this in general. But for testing purposes (in debug menu) not power-off resistant - why not? Amateur radio is experimental and if you cannot do something ugly by chance...

db4ple, 2019-08-14:
50 Mhz = 12.5 Mhz x 4 
The main problem is the problem of destroying the LPF by pushing too much energy through it.
Might not be  a real problem, the RD16HHF1 might not really deliver much power at 50 Mhz.

S52CQ, 2019-08-19:
Surely, nor RX or TX filters are not suitable for 52 MHz, nor is (unfortunately) RF output stage, which gives only 2W on 10m, while I guess it would deliver <1W at 52 MHz. So mcHF would need a heavy redesign to work on 52 MHz. Without any sun's activity upper bands are close so that does not make any sense. It would however, make sense to make 2m and/or 70 cm add-on PCB providing that oscillator can cover that.
73s de S52CQ.

closed 2019-08-20 by S52CQ