df8oe/UHSDR #1796: Will not boot up fully after upgrade

g7jiq, 2019-08-12:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Will not boot past first screen with ver number after update to latest 2.11.63 

## Issue Data goes here (please remove text above):

Your firmware version: 2.11.63
Your bootloader version: 5.0.4
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.x  early boards moded for touch screen with 512 mcu.
 * RF Board: mcHF 0.x 

### Describe the issue:

Fails to boot past first screen, 
First screen looks ok with no problems, and gos blank as normal.
But instead of the main radio screen all I get is the red led flashing, then the green led flashing then both flashing, then the green followed by the red etc.
(The amount of flash times changes as it cycles)
After cycling power and re-flashing to 2.11.62 all is well again.

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

df8oe, 2019-08-13:
Works for me... Maybe a damaged setting? If the latest fix (ver. 64) does not help you please try with "loading default configuration". Please leave a feedback if ver. 64 fixes your problem.

g7jiq, 2019-08-13:
Thanks for the latest fix,
All seems to work fine now, but will check out the full functionality later.

g7jiq, 2019-08-14:
Having tested as much as I can, all seems well.
Thanks

closed 2019-08-14 by g7jiq