df8oe/UHSDR #1798: Out Of Band TX

MatthiasDirlinger, 2019-08-18:
Your firmware version: 2.11.64
Your bootloader version: 5.0.4

### Hardware 
 * UI Board: mcHF 0.60
 * RF Board: mcHF 0.60

### Describe the issue:

Out Of Band the Power indicator is Red
On TX get Error: PA TX Power Factor is 0

### Your relevant settings

df8oe, 2019-08-18:
I can confirm. I never have had transmitted outside bands.

MatthiasDirlinger, 2019-08-18:
I tryed with 2.11.62 there its working

db4ple, 2019-08-18:
I will look into this.

MatthiasDirlinger, 2019-08-21:
My Solution here seems to work
Power not realy accurate out of band but i think this was before the same

// FIXME: This code needs fixing, the hack for TX Outside should at least reduce power factor for lower bands
    if (RadioManagement_IsGenericBand(band)) // we are outside a TX band
    {
        // TX outside bands **very dirty hack**
        //  FIXME: calculate based on 2 frequency points close the selected frequency, should be inter-/extrapolated
        uint32_t freq_min = RadioManagement_GetBandInfo(BAND_MODE_80)->tune;
        float32_t adj_min = ts.pwr_adj[ADJ_REF_PWR][BAND_MODE_80];

        uint32_t freq_max = RadioManagement_GetBandInfo(BAND_MODE_10)->tune;
        float32_t adj_max = ts.pwr_adj[ADJ_REF_PWR][BAND_MODE_10];

        float32_t delta_f = (float32_t)df.tune_old - (float32_t)freq_min; // we must convert to a signed type
        float32_t delta_points = freq_max - freq_min;

        float32_t freq_mult = delta_f / delta_points;

        pf_bandvalue =  freq_mult * (adj_max - adj_min) + adj_min;
    }
    else
    {
        pf_bandvalue = ts.pwr_adj[power == 0?ADJ_FULL_POWER:ADJ_REF_PWR][band->band_mode];
    }
    pf_bandvalue /= RadioManagement_IsPowerFactorReduce(df.tune_old)? 400: 100;

df8oe, 2019-08-21:
Does this reduce the output power outside bands to the recommended max. 50mW?

MatthiasDirlinger, 2019-08-21:
I dont think so but before your patch it has been the same
i found the limiting in RadioManagement_SetPowerLevel
there power is set to 50
but dont know where the final power calculation is done

df8oe, 2019-08-21:
Let's wait until Danilo takes a look at that. TX outside bands is a very low priority function (we are not allowed to TX outside bands and because of signal quality it is not recommended to use mcHF in front of transverters)...

roelandjansen, 2019-08-28:
>  (we are not allowed to TX outside bands and because of signal quality it is not recommended to use mcHF in front of transverters)...

does that apply only for OOB signals or the whole mcHF? I mean -- using 28...29.7 --> 2m for instance.

I rather would not see limits as there are sometimes reasons to have it TX outside of a band -- not into an antenna though.

df8oe, 2019-08-28:
We do not recommend to use mcHF for driving PAs or transverters because of signal purity reasons - please take a look at the recent discussions: https://www.amateurfunk-sulingen.de/forum/index.php?board=15;action=display;threadid=1413

closed 2019-10-24 by df8oe