df8oe/UHSDR #1799: Coupling Adj. on Higher Bands

MatthiasDirlinger, 2019-08-18:
Your firmware version: 2.11.64
Your bootloader version: 5.0.4

### Hardware 
 * UI Board: mcHF 0.60
 * RF Board: mcHF 0.60

### Describe the issue:
From 20m up in PA Configuration Menu always 40m is the only changeable value
I found no way to Calibrate 20m,15m,6m
for Example if a send on 15m the 40m Field goes changeable

### Your relevant settings

df8oe, 2019-08-18:
I can confirm. I have not recognized that because my adjustments are ready since years...

MatthiasDirlinger, 2019-08-19:
Works again
a 10m calibration point would be fine because my mchf does not transmit on 6m so i cannot calibrate bands over 15m

db4ple, 2019-08-19:
Well, I did not decide for the coupling ranges (they were like this for a long time). As you know it only affects the accuracy of the displayed "real" power. The 15m point is used to cover the range from 16 Mhz to 32 Mhz. 15m is roughly in the middle of this range. So it should be more or less okay across the range, at least that is the idea. If you for instance want it to be accurate as much as possible in the 10m band, just calibrate it so that it matches 10m.

MatthiasDirlinger, 2019-08-19:
Thanks i will try with 15m point and find a middle way

closed 2019-08-19 by MatthiasDirlinger