df8oe/UHSDR #1805: VSWR protection

S52CQ, 2019-09-01:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version: 2.11.62 and 2.11.67
Your bootloader version: 5.0.2
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.5
 * RF Board: mcHF 0.5

### Describe the issue:

Hi,

I was testing VSWR protection feature lately, and found out an issue there:
Turn VSWR protection ON;
control the radio via CAT;
whichever FW version mentioned above I use it causes TX LED (when controleld via CAT regardless which SW - tested was N1MM, WSJT-X)  to start flickering and no RF output is generated. If I press TUNE button set output power reverts to 0.5W (although It was set to 5W previously) and it all works well. But when controller with CAT LED flickers and there is not RF power.

73s de S52CQ, Jure

df8oe, 2019-10-11:
Please set VSWR protection to OFF (which is the default setting). Hardware for gathering VSWR information is not precise enough to get reliable functionality.

vy 73
Andreas

S52CQ, 2019-10-11:
OK, tnx.
73 de Jure

closed 2019-10-11 by S52CQ