df8oe/UHSDR #1806: TX filter OFF?

S52CQ, 2019-10-10:
Your firmware version: 2.11.67
Your bootloader version: 5.0.2
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.5
 * RF Board: mcHF 0.5

### Describe the issue:

Hi,
as I work with my mcHF quite ofter also FT8 I wonder if in digital signals any TX filter is used. namely at FT8 TX signal should be flat, while mine has slight AM modulation that can be seen - although at FT 8 there should be constant output RF.
I spoke with S52D Iztok and he told me that TX filter might be an issue. If it filters FT8 signal then it introduces distorsion and hence RF output is not flata, but slightly modulated - like at my mcHF.
So the question is: where can I turn OFF my TX filter? There are a couple of filters: BASS; TENOR; SOPRANO in the Filter Menu. AM filter can be turned OFF while SSB not.

73s de Jure

df8oe, 2019-10-11:
Hi Jure,

TX filter is a mandatory part of signal processing in IQ mixing concepts using digital hardware with reduced horse power. So (of course) there is NO setting "TX filter OFF". Tx'ing everything from 0Hz up to 2.8KHz would impact your signal quality impressively. Critical is the part from 0...~300Hz. This must be filtered as steep as possible.

Knowing this all the power of TX'ing FT8 is in your hands! You only have to move the signal into the range above 300Hz (I would recommend 500...1000Hz). FT8 is such a small signal: if you reside in the "working area" processing of such a small bandwidth *IS FLAT* of course :)

FT8 uses 8 tones at constant power so you need another signal purity: no distortions or intermixing. Signal quality of mcHF isn't the best so I recommend not to power up above 5W. That is the absolute maximum you should use: beter is 1 or 2W.

vy 73
Andreas

db4ple, 2019-10-11:
BTW, what you should turn completely off is TX Compression (CMP). This is probably responsible for what you called slight modulation. CMP 0 is not CMP OFF (!). Not that this would cause any real problems with FT8. 

While strictly seen this Audio compression is a TX filter, it just changes the signal amplitude. 

Same applies to BAS and TRB settings, but here 0 is off.

S52CQ, 2019-10-11:
OK, tnx. to both for the info.

73 de Jure

closed 2019-10-11 by S52CQ