df8oe/UHSDR #1807: ExtIO

sv1it, 2019-10-21:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:
Your bootloader version: 
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

sv1it, 2019-10-21:
Trying to operate my mchf with hdsdr I always run on the same error message : extIO not found , whenever I try to transmit. Receive works without any problems. My understanding was that mchf does not need an extIO.dll, but.......

Any ideas?

rogerclarkmelbourne, 2019-10-21:
You must fill in the details in the template.

Where do you see this message ?
What firmware version are you using
What hardware are you using
Etc etc etc

db4ple, 2019-10-22:
I truely believe this is a HDSDR configuration problem as the UHSDR firmware indeed does not need any specific ExtIO DLL. For the purpose of HDSDR your mcHF is a FT817 CAT controlled TRX producing/consuming IQ audio date connected via a standard USB soundcard interface.

So best is to ask some HDSDR cracks on the appropriate forums, this is not an issue of the UHSDR firmware.

closed 2019-10-24 by df8oe