df8oe/UHSDR #1811: RIT step size adjustment

rogerclarkmelbourne, 2019-10-25:
This is a feature request which I would like to implement, but would like some feedback before I start implementing it.

A friend of mine finds the RIT step size of 20Hz a bit too course when using CW and would like a lower setting.

I looked in the code and i can see that its hard coded to a step size of 20 and there is also a max number of steps.

https://github.com/df8oe/UHSDR/blob/37197a359af000688b17482c68bb1a081ff8d270/mchf-eclipse/drivers/ui/radio_management.c#L593

https://github.com/df8oe/UHSDR/blob/37197a359af000688b17482c68bb1a081ff8d270/mchf-eclipse/drivers/ui/radio_management.c#L812

https://github.com/df8oe/UHSDR/blob/9af8ab68d3d9a2e7f1bccfddf023e45434957083/mchf-eclipse/drivers/ui/ui_configuration.h#L67-L68

So potentially all of these would need to be changed.

However, 10Hz or less step size for RIT may be too fine for other modes, and perhaps it would be better if this was implemented as a setting.

And if its a setting, then perhaps the setting should be per mode, i.e a different setting for CW to SSB, AM and DV.  (LSB and USB etc would just be grouped as the SSB setting)

I have not looked at the way that the settings are handled, but hopefully i could add a new group of settings and also add something into the UI to allow the setting to be adjusted.

db4ple, 2019-10-26:
Adding configuration settings controlled via menu is relatively straight forward:

ui_menu.c / ui_menu_structure.c

And for persistent storage use: ui_configuration.h/.c
It is possible to store 2 8bit values in a single config value record (or one 8/16/(32) bit value.)

rogerclarkmelbourne, 2019-10-26:
Thanks.
I already recompiled with 10Hz step size and limit of +/- 99 steps 

I will look at the menu and storage for this,

Also. Control could perhaps somehow be via press on RIT encoder.