df8oe/UHSDR #1812: Holding down the RIT button causes unexpected behaviour

rogerclarkmelbourne, 2019-10-25:
Your firmware version: V2.11.67 

Holding down the RIT button seems to cause the associated setting e.g. "Mic" to change

Note. I was investigating whether I could add new functionality to adjust the RIT step size, by holding down the RIT button and adjusting the RIT control, (see my feature request consultation issue https://github.com/df8oe/UHSDR/issues/1811 )
But I've not changed any code yet.

db4ple, 2019-10-26:
This is absolutely expected behavior. Pressing one of the encoder buttons is technically pressing M1 to M3:
https://github.com/df8oe/UHSDR/wiki/Quick-Operating-Guide

Like in the mcHF 0.7 the manufacturer merged the separate buttons into the encoders. 
So unless you want to change the whole UI concept, this has to remain as is. 
BTW, the OVI40 hardware has separate buttons in the encoders (which the firmware does not support yet). Not that this information is going to help in this case...

db4ple, 2019-10-26:
Forgot: Details regarding buttons: https://github.com/df8oe/UHSDR/wiki/Operating-Manual

rogerclarkmelbourne, 2019-10-26:
OK.

Thanks.

closed 2019-10-26 by rogerclarkmelbourne

rogerclarkmelbourne, 2019-10-26:
Perhaps this is not documented very well.

If RIT is the active input, pressing and holding the RIT button or the RIT encoder does not change the active item from RIT control to L>L / L>R etc. 

It changes the Inactive element L>L / L/R even though they are not show as active, which seems strange that an inactive UI element is being changed.

db4ple, 2019-10-26:
Well, the wiki table in https://github.com/df8oe/UHSDR/wiki/Operating-Manual clearly describes "long press" of M3 (which is on the RS 918 and mcHF 0.7 the RIT encoder button) will switch "line in channel". 
A short press will change the active item and the table says "Encoder 3 value". Maybe we have a better wording for this.  
You are welcome to improve the documentation.