df8oe/UHSDR #1827: SI569 instead of SI570

S52CQ, 2019-11-22:
It's not really an issue but more a question if any of you guys have thought of replacing SI570 with similar oscillator available from the same manufacturer - SI569 that appears to be a new chip at Silicon Labs, it's freq. range is from 200 kHz to 3 GHz, it's got lower jitter and it's pin compatible with SI570.

73s de S52CQ, Jure

df8oe, 2019-11-22:
Not yet. All kinds of SI569 which are available cost 60 Euros or more, and all other are not available. If someone can contribute such an oscillator (the version which goeas up to 1.8G is sufficient) we could start investigating :) It would be interesting, because it goes down to 200KHz (which would result to a working frequency of 50KHz) and possibly can handle 2m, too (very, very, very, very, very, very, very, very experimental of course)...

73 de
Andreas

S52CQ, 2019-11-25:
Hi Andreas, yes, they are quite expensive, but offering a lot in that little package :) 

I do happen to know some people in industry, most likely will meet them at Embedded World next Feb. in Nurnberg. Perhaps I can get a sample or two and would gladly send them to you for a test. Which particular part No. would suit you best? 569AAAA or 569AAAB or even 569AAAC? Last 4 letters should probably be ABGR, we do not want to have 3.2 x 5 mm package right? 

73 de S52CQ, Jure

df8oe, 2019-11-25:
I would prefer AAAC. If set is build using potatoe high speed chips possibly 2m can be used (if doing a lot of external filtering). I do not think 70cm will work due to layout - so AAAB is not neccessary. Regarding package I agree to your opinion.

73 de
Andreas

db4ple, 2020-10-04:
Hi, I know this is a little old, but anyways: the Si569 is the VCXO, the actual replacement for the Si570 seems to be the Si564. But as far as I know we  have to use the CMOS version, and this is rated (same for Si569 and Si564 btw) for up to 250 Mhz only. This is still an improvement from the Si570 CMOS which was rated for 160Mhz (but work in the MCHF with up to 220Mhz). So 2m is out of reach, as this would require close to 600 Mhz upper limit.  But 4m could be doable if we have a similar additional range as with the Si570 (+60 Mhz to official limit).

S52CQ, 2020-10-05:
Hi,

thanks for that. Well it does not matter much now. At that time I was 
asking some guys from distribution to get me some samples, but it was no 
reply so no samples from my end I guess.

Best 73s de S52CQ, Jure

db4ple, 2021-02-19:
I think we can close this now.

closed 2021-02-19 by db4ple

srgmn, 2024-09-29:
Estive a procura do SI570 e não consegui encontrar, está fora do mercado já. Alguém tem uma sugestão de algum substituto pare o SI570 no ano de 2024?