df8oe/UHSDR #1828: Artefacts on screen when switching fro SPLIT back to normal operation

OK1FWG, 2019-12-15:
# UHSDR Firmware Defect Issue (Bug) Template 1.1

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version: 2.11.75
Your bootloader version: 5.0.4
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.7
 * RF Board: mcHF 0.7

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

If you have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes
 * Magnify: 4x 
 * Spectrum: Dual
 

---> When switching from SPLIT operation back to normal, there are artefacts from the 'split' screen. It seem to me like new screen refresh is needed when switching to normal display to erase previous information from the buffer. It is directly behind the figures displaying the actual frequency. (If neede, I can provide a photos).

73, Willi, OK1FWG

df8oe, 2019-12-16:
Hi Willi,
thanks for the report. I can reproduce - but only in one combination:
- 3.5" LCD @480x320
- AND "modern font"

Bug has low priority.

vy 73
Andreas

db4ple, 2020-02-11:
@OK1FWG: Hi, did we fix this? Please verify and if yes, close the issue.

db4ple, 2021-02-19:
Closed due to inactivity

closed 2021-02-19 by db4ple