df8oe/UHSDR #1831: wrong RAM size in System Info with 480MHz STM32H743ZIT6U

DL8EBD, 2020-01-10:
Firmware:  D2.22.78
Bootloader: 5.0.4
Hardware: SParrow 

**only a minor bug:**
with the new H7 CPU STM32H743ZIT6U, we see a wrong RAM size in system Info

[image]

73
Thomas

db4ple, 2020-01-10:
Well, it depends what is the definition here: We show the "usable" RAM for UHSDR, i.e. not necessarily the full RAM provided by the hardware. And on the H7 we indeed use "only" 512k. The RAM is scattered (not my words, these are STM's words: https://www.st.com/en/microcontrollers-microprocessors/stm32h743-753.html ).
So it is 64 ITCM + 128 DTCM + 512 + 288 + 64 user RAM and finally 4k Backup RAM. This adds up to 1024k. We are just using the large 512k block of it.

Not sure if there is a need to fix this. Maybe we adds this to the help text.

DL8EBD, 2020-01-10:
Thanks Danilo!
I think we can close this issue.
There is no bug and the question has been answered.

73
Thomas

closed 2020-01-10 by DL8EBD