df8oe/UHSDR #1832: Enhancement

co6egm, 2020-01-23:
##  Reverse scrolling outside menu:

Your firmware version: 2.11.78
Your bootloader version: 5.0.4

### Hardware 
 * UI Board: mcHF 0.6.3
 * RF Board: mcHF 0.6.3

Is posible to add a reverse scrolling option related to the AFG/AGC/RIT options on the configuration menus? The issue apears at the time I replace my supplied  encoders for a knurled shaft ones... It start to increase AFG, AGC and RIT respectively when I turn the encoders counterclockwise. It was increasing when turning clockwise before this mod.

df8oe, 2020-01-23:
It is possible - but is it useful to add everything that is possible? We have talked about this some time before.

Menu of UHSDR is alreeady very blown up. There are a few options we want to think about. Adding new options which are only targeted to a handful users is nothing we want to do. > 99% of all users use encoders that are connected correctly and those users will be bothered by such a new option. So I recommend to cut the tracks on your device and cross the encoder lines. Encoders are not "pluggable" - so a software solution with user defineable setting is not the user friendly way

co6egm, 2020-01-23:
Hello there and thanks for your reply.
I think this add on can get an open range of option to the user, remember
not everyone wants to get a partially assembled kit or a finished product,
and encoders are hard to remove without induce damage on the logic board. I
don’t think the “cut the track” solution is the better (but quicker) idea.
On the other hand I’ll try to do the mod my myself, but I’m not a
programmer. I really enjoy to assemble kits and build rigs from scratch...

rogerclarkmelbourne, 2020-01-23:
@co6egm 

I you need to remove the encoders, I recommend you cut the legs and then removed the encoder. Then remove the lower parts of the legs from the PCB, as this will cause minimal damage.

New encoders are cheap. They are around 40c each if you buy a pack of 5 from China etc

co6egm, 2020-01-23:
@Roger Clark
I’m that case which encoder you recommend??? The ones I bought was the EC11
model, but it works oposite compared with the encoders that Chris and Lucy
sent with the kit.
My goal is make it look pretty like it looks now.

rogerclarkmelbourne, 2020-01-24:
@co6egm 

My encoders look like EC11, so I'm not sure why yours go in the opposite direction.

You could just build your own custom version of the firmware, with the encoder directions reversed.
But I appreciate it could be a problem if any code in the same file, ever changes in the master version of the repo, as you would have to manually update your code.

df8oe, 2020-01-24:
That is the reason I do not recommend to use encoders with reversed direction. Or, if you do it, modify the PCB instead of the firmware. You will have to build your own version every time a new firmware is released. You are leyving the supported strack if you use such encoders without any modification... Same for using another oscillator chip. There are dozens which will work, too and may have advantages - but they are not officially supported.

co6egm, 2020-01-24:
I agree with you... for a very long toma I used the SI5351A on my projects.
It works fine but it’s a little cheap and dirty solution. The better
solution in the big brother SI570...
I will probably try to get another knobs that fir on the original encoders
because those encoders works as expected.
A lot of Chinese sellers have pretty good deals on encoders and another
stuff, but recently I checked 3 small orders that y made in the last fit
the same model, and I found just one that match with the original encoders,
and the other two orders works opposite. But those encoders are all D-shape
shaft...

df8oe, 2020-01-24:
PCB is only two layers. So it is not as complicated to cut the two outer tracks using a cutter knife and rewire them using some thin wires. The absolute best solution would have been zero ohms SMD resistors in layout - but that is "too late". You only have to do the rewiring one time and if you open the case to get access to the PCB it does not take longer to do the modification to all needed tracks than changing the firmware. But you have to change the firmware each time a new version appears and only once if you "hard-code" it to the PCB ;)

co6egm, 2020-01-24:
I already fixed the rig...
now i have to dig on my 3d printer code because it’s acting weird...
A few years ago I was studying to become an electrical engineer, and I had
a programming class. I used to build all my programs using the same
pattern... I used a block pattern to easily add more functions, menus, etc
etc. It also make it easy to modify the available options.
But yes, like you said the menus on the mcHF program are blown up, and it
doesn’t mean the user have to be an expert to calibrate the rig, but
definitely I as the user need to get the instructions on a hard copy or on
my bench computer to follow the steps...

closed 2021-02-19 by db4ple