df8oe/UHSDR #1837: About Operating Manual on Wiki

satoryboy, 2020-03-02:
Your firmware version: 2.11.79
Your bootloader version: 5.0.4
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.6 (R928+ with Si570)
 * RF Board: mcHF 0.6 (R928+)

### Describe the issue:
Please supplement Operating Manual on Wiki:

In the "Key Press Map" table, a long press of BAND- or BAND+: jump down / up by frequency 
to the width of the panorama.

In the table "Color and Blink Codes of the LEDs Explained": frequent flashing of the red LED - protection operation by SWR.

df8oe, 2020-03-02:
No need to open an issue - you can correct / add this by yourself!

closed 2020-03-02 by df8oe

satoryboy, 2020-03-02:
OK! How?

satoryboy, 2020-03-02:
Good day, Andreas. 
I did not find any instructions on how to edit the wiki of the project. I hope for your help. Enough links.

> 
>Closed  #1837 .

db4ple, 2020-03-02:
In DF8OE Github: just click on Wiki and go to page you want to change. If you are logged into your github account, you will be able to see an EDIT button in top part of the page.

satoryboy, 2020-03-02:
Thank you very much! 73!

> 
>In DF8OE Github: just click on Wiki and go to page you want to change. If you are logged into your github account, you will be able to see an EDIT button in top part of the page.