df8oe/UHSDR #1838: update for Wiki

csylvain, 2020-03-03:
i have made the UHSDR wiki use visual media stored within it's own repo (for maintainability) and have incorporated material from EA3HZZ which provides connector pinout information in the new "mcHF Connections" page.

i would have submitted this update as a pull request, except github does not offer than for a wiki subordinate to a code repo.

to update the df8oe/UHSDR.wiki it will be necessary to clone https://github.com/csylvain/UHSDR.wiki.git locally, then push to df8oe/UHSDR.wiki

df8oe, 2020-03-03:
The idea to put WIKI images in repo is a nice one. Before cloning the WIKI and pushing it I must investigate how to see the changes - I never have done such a work on WIKI ;)

csylvain, 2020-03-03:
what you will get is what is in my wiki's repo right now.  git pull your
own wiki, then git pull my wiki, then copy everything from mine on top of
your local copy of your wiki.  then you can work git like normal in your
wiki's copy.

csylvain, 2020-03-03:
sorry "git clone"  not "git pull"   :-P

df8oe, 2020-03-04:
I am nearly sure there is a simple way to see the diffs like in "normal git". But this week I am very busy and so I will start this at the weekend. And of course I will publish my "way".

csylvain, 2020-03-04:
if we could fork a wiki just the same as a code repo, then it would be
really simple. unfortunately, in github world, wiki is still a different
kind of repo than a code repo. i do know if you cloned them both locally
and just overwrote the cloned original content files with the new content
files, then you'd be able to see the changes in a normal way like you had
done it all yourself. then you need only choose to commit or revert or edit
further ...

df8oe, 2020-03-06:
Hi,
I have got it work but I am not sure everything is "as expected"... Firstly I have got information that wiki can be handled same as code area. So I added an upstream:
git remote add upstream https://github.com/csylvain/UHSDR.wiki.git
Next I fetched the contents:
git fetch upstream
and merged it into my local copy:
git merge upstream/master
No errors popped up - and there was a new folder created in wiki: images... Now I tested locally if everything works and it did. So I pushed it with:
git push
Only thing where I am not clear: The folder containing the images now is located within wiki area. I do not think taht is soooo bad: it IS wiki contents, and if it can be handled with standard git commands: why not? You can maintain it easily via git...

csylvain, 2020-03-06:
what i mean by the wiki repo not being exactly the same as a code repo is
you do not have the option to work with it the same way via a web browser
at github.com.  there also is not the same mechanism for creating a pull
request via a browser at github.com.  yes, as long as you use the 'magic'
wiki repo URL, then you can manipulate repos on your local drive like
normal. for now, you must rely on your local storage to work with wiki
repos where github makes it more easy when working with a code repo.

also, until recently, and according to some people who wrote about
developing a github wiki, 'magic' URLs were needed in the Markdown code to
reference binary blobs like images. fortunately, there is now some simple
MediaWiki code which can be written.

it is certainly desirable and recommended to store wiki media within the
wiki itself. there is always a risk of media being unavailable when
referencing media external to github when github is the host of the wiki.

i hope we are both now in complete understanding of each other.