df8oe/UHSDR #1841: CW iambic keyer issues

co6egm, 2020-04-08:
Your firmware version: 2.11.79 and 2.11.78
Your bootloader version: last one
 * UI Board: mcHF 0.6.3
 * RF Board: mcHF 0.6.3
Unable to get dits and dashses when using an iambic keyer. Attempt to reverse keyer on firmware but only dits or dashes, not both. When paddles are squeezed, runs as expected.
I also tried downgrade firmware to v2.11.78 but still the same.

df8oe, 2020-04-08:
I cannot confirm. Here everything runs as expected. I hope you do not have connected an external iambic keyer. If so: you must use streight key mode. If you have connected paddles you maybe you have a connection issue to the plug.

co6egm, 2020-04-08:
Fixed... I found a cold joint on the pcb

df8oe, 2020-04-08:
Very good! Please close this issue now.

vy 73
Andreas

closed 2020-04-08 by co6egm